openwrt/luci #4868: luci-app-banip: Empty string warning

feckert, 2021-03-03:
I get the following error on I18N update.

```
Updating applications/luci-app-banip/po/templates/banip.pot ... standard input:126: warning: Empty msgid.  It is reserved by GNU gettext:
                             gettext("") returns the header entry with
                             meta information, not the empty string.
done
```

I think this [line](https://github.com/openwrt/luci/blob/master/applications/luci-app-banip/htdocs/luci-static/resources/view/banip/ipsetreport.js#L126) is the root cause of this mesage. If I remove the line the warning disappears. But I am not familiar with your code, so could you @dibdot please check if it is save to remove the line to fix this warning?

dibdot, 2021-03-03:
The table row is needed, just remove the translation and fill in a non breaking space instead (see #4869).

feckert, 2021-03-03:
Fixed with https://github.com/openwrt/luci/pull/4869

closed 2021-03-03 by feckert