openwrt/luci #4870: Port Forwarding Non-Contiguous Source and Destination Ports

lwbeam, 2021-03-03:
Currently the user is only allowed to specify a single port, or range of contiguous ports, when configuring port forwarding using the OpenWrt/LuCI (19.07.7) interface (Network > Firewall > Port Forwarding > Add). As a result, the OpenWrt/LuCI interface does not allow the user to specify a list of non-contiguous ports. For example, when setting up port forwarding for VNC, the user is not allowed to specify ports 5500, 5800 and 5900 as a list in a single entry.

Furthermore, this appears to only be a limitation of the OpenWrt/LuCI interface, as a list of non-contiguous ports can be specified by either editing the /etc/config/firewall file or using the following UCI commands:

uci set firewall.@redirect[-1].src_dport="5500"
uci add_list firewall.@redirect[-1].src_dport="5800"
uci add_list firewall.@redirect[-1].src_dport="5900"

Finally, once configured, the OpenWrt/LuCI interface (Network > Firewall > Port Forwarding) correctly displays these non-contiguous ports.

Please update the OpenWrt/LuCI interface to allow lists of non-contiguous source and destination ports to be specified when setting up Port Forwarding (Network > Firewall > Port Forwarding > Add). Many thanks.

jow-, 2021-03-03:
The underlying firewall3 does not support this:

```
root@jj:~# uci show firewall.@redirect[11]
firewall.cfg203837=redirect
firewall.cfg203837.name='Testing'
firewall.cfg203837.src='wan'
firewall.cfg203837.proto='tcp'
firewall.cfg203837.src_dport='5500' '5600' '5700'
firewall.cfg203837.dest_port='1234'
firewall.cfg203837.target='DNAT'
root@jj:~# fw3 reload
Warning: Option @redirect[11].src_dport must not be a list
Warning: Section @redirect[11] (Testing) skipped due to invalid options
```

closed 2021-03-03 by jow-

lwbeam, 2021-03-03:
With OpenWrt/LuCI (19.07.7) installed:

root@OpenWrt:~# uci show firewall.@redirect[1]
firewall.cfg103837=redirect
firewall.cfg103837.dest_port='5500 5800 5900'
firewall.cfg103837.src='wan'
firewall.cfg103837.name='iMac (VNC)'
firewall.cfg103837.src_dport='5500 5800 5900'
firewall.cfg103837.target='DNAT'
firewall.cfg103837.dest_ip='192.168.0.189'
firewall.cfg103837.dest='lan'

My mileage differs. It may require that the source and destination lists are the identical, which would make sense to me.

jow-, 2021-03-03:
Does not work either:

```
root@jj:~# uci show firewall.@redirect[11]
firewall.cfg203837=redirect
firewall.cfg203837.name='Testing'
firewall.cfg203837.src='wan'
firewall.cfg203837.proto='tcp'
firewall.cfg203837.src_dport='5500 5800 5900'
firewall.cfg203837.dest_port='5500 5800 5900'
firewall.cfg203837.target='DNAT'
root@jj:~# fw3 reload
Warning: Option @redirect[11].src_dport has invalid value '5500 5800 5900'
Warning: Option @redirect[11].dest_port has invalid value '5500 5800 5900'
Warning: Section @redirect[11] (Testing) skipped due to invalid options
[...]
```

lwbeam, 2021-03-03:
Thanks for taking the time to look at it. It's greatly appreciated.