openwrt/luci #4872: Wireless Channel Selection

lwbeam, 2021-03-03:
OpenWrt/LuCI (19.07.7) uses the [CRDA database](https://openwrt.org/docs/guide-user/network/wifi/wifi_countrycode) to determine what wireless radio settings and channels are legally allowed to be used in each country. When configuring a wireless radio within OpenWrt/LuCI (Network > Wireless > Edit > Device Configuration / General Setup > Channel), this results in the user being presented with a complete list of all legally allowed channels for the specified country. Given my limited _testing_ for my country code, this appears to work correctly.

However, for the 26 5GHz channels presented by OpenWrt/LuCI for country code "CA", only nine actually work on my D-Link DIR-885L A2 router. Specifically, these are Channels 36, 40, 44, 48, 149, 153, 157, 161 and 165. Furthermore, these channels are the only 5GHz channels allowed by the OEM firmware, which likely represents a limitation of the underlying hardware. As additional support for this, the identical channel list appears to be embedded in the OEM firmware for [Linksys EA9500](https://forum.dd-wrt.com/phpBB2/viewtopic.php?p=1063474), which uses the same Broadcom chip.

As OpenWrt/LuCI is meant for use with a great many different Makes and Models of hardware, it would be unrealistic to expect it to automatically limit the radio channel selection to only those supported by the hardware on which it has been installed. However, it would be extremely useful to be able to limit the channel selection to only those supported, when this is know. To this end, I'd like to suggest the addition of a section/option to the /etc/config/luci file, that would allow the user to specify a list of supported channels. This could be done as a single list, or separate lists for 2.4 and 5GHz channels. OpenWrt/LuCI would then present only those channels from the list(s) that are legally allowed by the specified country code. If no list is provided, all legally allowed channels would be presented. Appropriate UCI support would also need to be added for this new config file section/option. Many thanks.

nickberry17, 2021-04-01:
I haven't checked this, but what channels does the `iw` command show the radio being capable of?

adelton, 2025-01-11:
@lwbeam Could you provide the `iw` information requested by ghost?

systemcrash, 2025-01-11:
Not a luci bug

closed 2025-01-11 by systemcrash