openwrt/luci #4879: luci-app-radicale2: Can't open Authentication / Users nor Storage tab

ikazuhiro, 2021-03-07:
I use locally built master branch.  `Server settings` and `Logging` tab have no problem in my router's radicale2 page, .  But when I open `Authentication / Users` or `Storage` tab, an error like below occured.

> /usr/lib/lua/luci/model/cbi/radicale2/auth.lua:3: attempt to index field 'controller' (a nil value)
> stack traceback:
> 	/usr/lib/lua/luci/model/cbi/radicale2/auth.lua:3: in function 'func'
> 	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
> 	/usr/lib/lua/luci/dispatcher.lua:1353: in function '_cbi'
> 	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
> 	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

Modifying
`
local rad2 = luci.controller.radicale2
`
in `auth.lua` and `storage.lua` to
`
local rad2 = require "luci.controller.radicale2"
`
like `server.lua` seems to fix this issue, but I'm not sure this is correct fix.

ddast, 2021-09-20:
I am still seeing this issue in OpenWrt 21.02.0.  I also checked that the proposed modifications fix the issue.

Do you want to create a pull request for this or are you fine with me creating one to get this fix integrated?

ikazuhiro, 2021-09-21:
I don't have enough time, so please go ahead if you would.