openwrt/luci #4881: Lower `apply_holdoff` time

aparcar, 2021-03-07:
From my observation *applying* changes via LuCI takes nearly always less than 5 seconds, except maybe for changes that e.g. change interfaces or WiFi options. I briefly checked the code and I suspect [the view header definition](https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/view/header.htm) `apply_holdoff` to always add some wait time.

Couldn't we make the interface "more snappy" by setting the holdoff to 1 second but then increase it to 5 seconds on second try?

hnyman, 2021-03-07:
Hmm... Trying to remember the original discussions from 2018.

I think that it was originally shorter and that caused problems with some apps, so it was made a bit longer (and configurable)

Discussion e.g. here in the original topic about the "new" rollback: https://github.com/openwrt/luci/pull/1769#issuecomment-386068656

adelton, 2025-01-11:
Given this is now configurable, is there anything else that needs to be investigated, or can this issue be closed?