openwrt/luci #4894: luci-app-statistics: ping plugin does not show graphs

dengqf6, 2021-03-09:
[image]

hnyman, 2021-03-09:
Sorry, I have never used nginx.
But it is really hard to believe that the webserver in use would have impact.

All graphs are generated in pretty standard way from the rrd data. There are image recipes for each stats item, but the recipes are quite similar. If the ping graphs fail, then likely also all other graphs should fail with nginx.

I would instead look into your data collection. Does the ping actually store stats data? Are there rrd databases?

```
 OpenWrt SNAPSHOT, r16152-bff84f3e8e
 -----------------------------------------------------
root@router1:~# ls /tmp/rrd/router1/
conntrack                load                     thermal-thermal_zone3
cpu-0                    memory                   thermal-thermal_zone4
cpu-1                    ping                     thermal-thermal_zone5
cpufreq-0                thermal-cooling_device2  thermal-thermal_zone6
cpufreq-1                thermal-cooling_device3  thermal-thermal_zone7
entropy                  thermal-thermal_zone0    thermal-thermal_zone8
interface-br-lan         thermal-thermal_zone1    thermal-thermal_zone9
iwinfo-wlan0             thermal-thermal_zone10
iwinfo-wlan1             thermal-thermal_zone2

root@router1:~# ls -l /tmp/rrd/router1/ping/
-rw-r--r--    1 root     root          7096 Mar  9 17:07 ping-87.92.0.1.rrd
-rw-r--r--    1 root     root          7096 Mar  9 17:07 ping_droprate-87.92.0.1.rrd
-rw-r--r--    1 root     root          7096 Mar  9 17:07 ping_stddev-87.92.0.1.rrd

```

use rrdtool dump to examine the data.

```
root@router1:~# rrdtool dump /tmp/rrd/router1/ping/ping-87.92.0.1.rrd | less
<!-- Round Robin Database Dump -->
<rrd>
        <version> 0001 </version>
        <step> 30 </step> <!-- Seconds -->
        <lastupdate> 1615302378 </lastupdate> <!-- 2021-03-09 17:06:18 EET -->

        <ds>
                <name> value </name>
                <type> GAUGE </type>
                <minimal_heartbeat> 60 </minimal_heartbeat>
                <min> 0.0000000000e+00 </min>
                <max> 6.5535000000e+04 </max>

                <!-- PDP Status -->
                <last_ds> UNKN </last_ds>
                <value> 2.2176000000e+01 </value>
                <unknown_sec> 0 </unknown_sec>
        </ds>

<!-- Round Robin Archives -->
        <rra>
                <cf> AVERAGE </cf>
                <pdp_per_row> 1 </pdp_per_row> <!-- 30 seconds -->
                <xff> 1.0000000000e-01 </xff>

                <cdp_prep>
                        <ds><value> NaN </value>  <unknown_datapoints> 0 </unknown_datapoints></ds>
                </cdp_prep>
                <database>
                        <!-- 2021-03-09 15:54:30 EET / 1615298070 --> <row><v> 1.0944000000e+00 </v></row>
                        <!-- 2021-03-09 15:55:00 EET / 1615298100 --> <row><v> 1.0874000000e+00 </v></row>
                        <!-- 2021-03-09 15:55:30 EET / 1615298130 --> <row><v> 1.0944000000e+00 </v></row>
                        <!-- 2021-03-09 15:56:00 EET / 1615298160 --> <row><v> 1.0930000000e+00 </v></row>
                        <!-- 2021-03-09 15:56:30 EET / 1615298190 --> <row><v> 1.1128000000e+00 </v></row>
                        <!-- 2021-03-09 15:57:00 EET / 1615298220 --> <row><v> 1.1310000000e+00 </v></row>
```

Other idea might be that something in the filename parsing goes awry?  IPv6 address (with : ) to be pinged? Illegal chars in hostname?
Or something like that.

hnyman, 2021-03-09:
I wasn't sure if that #4358 was right , so I assigned it to @jow- who had merged the corresponding fix to 19.07
He has apparently not yet looked into it.

Ps. You might have mentioned that ipv6 in your original issue report, as you have had trouble with that also earlier.

dengqf6, 2021-03-10:
[image]
I tried using hostnames instead but luci does not like it. Yet another bug? @jow-

hnyman, 2021-03-24:
Likely fixed by #4358

closed 2021-03-24 by hnyman