openwrt/luci #4911: Shadowsocks is no longer safe, do you consider supporting shadowsocksr+?

i765372386, 2021-03-11:
Many people like the official safe and concise installation image, but there is no luci -app-shadowsocksrplus in the maintained application, so many people have to compile other images. As shadowsocks is no longer safe, do you consider supporting shadowsocksr+?

feckert, 2021-03-11:
Feel free to add this support.

dengqf6, 2021-03-11:
As Shadowsocksr is an unofficial fork of Shadowsocks, and remains unmaintained [for years](https://github.com/shadowsocksr-backup/shadowsocksr-libev), why do you get the feeling it is more secure?

openwrt-diy, 2021-03-13:
@dengqf6 
The poster mentioned that shadowsocksr+ refers to luci-app-ssr-plus! It is widely used in China and overseas Chinese regions. It has existed for two or three years and the source code is always being upgraded. luci-app-ssr-plus also supports the official version of OpenWrt, the only difference is that the LUCI interface support is not friendly enough!

Source code maintainer @coolsnowwolf
Source code address
https://github.com/fw876/helloworld/tree/master/luci-app-ssr-plus

closed 2021-03-13 by i765372386