openwrt/luci #4922: read-only filesystem check

coelner, 2021-03-14:
I had problems to save a changed configuration and it takes some time to find the curiosity: the overlay filesystem was mounted readonly, but I need to check this over ssh.
However a short message in the right corner like the field "unsaved changes" would be nice. 
Even the mount points page within luci didn't show the readonly status and therefore did not show a button to remount this as readwrite.

dangowrt, 2021-03-14:
Was it actually overlayfs being mounted r/o or rather OpenWrt's preinit deciding to use overlayfs with upperdir==tmpfs due to other options being unavailable?
I'm asking this because preinit should make sure we NEVER end up with read-only rootfs and in case `rootfs_data` being unavailable it should use tmpfs instead. I could create an additional attribute in `ubus call system board` to allow detecting that state more comfortably (if that's actually what you are observing).

coelner, 2021-03-14:
No, the overlayfs was mounted r/o. I don't know why this happened. I rebooted the device, but this didn't happened again.
```
root@openwrt:~# mount
proc on /proc type proc (ro,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (ro,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (ro,relatime,mode=600,ptmxmode=000)
mountd(pid2330) on /tmp/run/blockd type autofs (ro,relatime,fd=7,pgrp=1,timeout=30,minproto=5,maxproto=5,indirect)
[reboot]
root@openwrt:~# mount
/dev/root on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mtdblock4 on /overlay type jffs2 (rw,noatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
mountd(pid2192) on /tmp/run/blockd type autofs (rw,relatime,fd=7,pgrp=1,timeout=30,minproto=5,maxproto=5,indirect)
``` 
I guess that something went horrible wrong. I can't reproduce that issue, but nevertheless, maybe a warning symbol would be useful

dangowrt, 2021-03-14:
Hm, maybe the filesystem holding the upperdir got corrupted and was automatically re-mounted read-only, and that change then propagated to overlayfs.

coelner, 2021-03-14:
Sorry, no. I do not have any further information about this

hnyman, 2021-03-14:
> because preinit should make sure we NEVER end up with read-only rootfs and in case rootfs_data being unavailable it should use tmpfs instead. 

I think that there is an exception: if the overlay exists but is too full (preventing full overlayfs working), overlayfs mounts it as read-only.

Refernce to:
https://github.com/openwrt/openwrt/commit/18691109df5a590e9914ae73740f145beec847ce

coelner, 2021-03-14:
That should not be a problem here:
```
root@openwrt~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 8.3M      8.3M         0 100% /rom
tmpfs                    60.7M    148.0K     60.6M   0% /tmp
/dev/mtdblock4            6.1M    468.0K      5.7M   7% /overlay
overlayfs:/overlay        6.1M    468.0K      5.7M   7% /
tmpfs                   512.0K         0    512.0K   0% /dev
```