openwrt/luci #4926: luci write incorrect service when "--custom--" is chosen

wonson, 2021-03-16:
luci-ddns write incorrect service into /etc/config/ddns when "--custom--" service is chosen
`option service_name '-'` where this line should be deleted or commented ( https://openwrt.org/docs/guide-user/services/ddns/client#custom_service1 )

This would result in "No update_url found/defined or no update_script found/defined! - TERMINATE" error when it try to update

tavinus, 2022-03-31:
I was having a look at the code and I think your solution above is not ideal. From what I can understand, we need to suppress the `service_name='-'` **ONLY** when saving the configuration file.  
  
So, that is when we call the `uci.set` command on `service_name`  
[Here](https://github.com/openwrt/luci/blob/master/applications/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js#L650)
```javascript
uci.set('ddns',section_id,'service_name',service_value);
```

As it is already done in other parts of the code, we need to check if it is a custom script and then IGNORE the uci.set command for the service_name.  
  
We can do that with
```javascript
if (service_value != '-') { uci.set('ddns',section_id,'service_name',service_value); }
```
  
Was quite the pain to test this in a running OpenWRT router. Browser cache also interfering.  
  
Also, this is done when you select the service type and click `CREATE SERVICE` (pre-creation)
  
You can patch the minified version like I did if you want.   
  
File: `/www/luci-static/resources/view/ddns/overview.js`  
  
Original:
```javascript
return m.save(function(){uci.add('ddns','service',section_id);uci.set('ddns',section_id,'service_name',service_value);uci.set('ddns',section_id,'use_ipv6',ipv6_value);})
```
  
Changed:
```javascript
return m.save(function(){uci.add('ddns','service',section_id);if(service_value!='-'){uci.set('ddns',section_id,'service_name',service_value);}uci.set('ddns',section_id,'use_ipv6',ipv6_value);})
```

Then reload  uhttp
```
/etc/init.d/uhttpd reload
```

Your browser may be stubborn about the Javascript cache though. Try in another browser to be sure (or inspect the source).  
  
I think this is a simple working solution. I probably won't be able to make a PR any time soon.  
  
  
Cheers!  
  
Gus

tavinus, 2022-04-01:
Sorry to double-post, but this is something else kind of related.  
  
I noticed that when creating the custom service, you will have to select some other service and then select the `-- custom --` service back for it to work. I think there used to be ANOTHER first option with a description before the custom one (am I wrong?). This may be related to removing that? Maybe some code that used to compensate for that?  
  
This is how it behaves
[image]

Then you click something else
[image]

Then `-- custom --` works
[image]

attila-lendvai, 2022-04-03:
FWIW, i was bitten by this just now.

commit 85ff9db2 (2022-10-11): luci-app-ddns: fix creation of custom service (fix #4926)

The `service_name` option must be empty for custom services,
otherwise an error is raised in `dynamic_dns_updater.sh`.
commit b139f13c (2022-10-11): Merge pull request #5999 from jcassette/issue-4926

luci-app-ddns: fix creation of custom service (fix #4926)
commit 938bb8f7 (2022-10-12): luci-app-ddns: fix creation of custom service (fix #4926)

The `service_name` option must be empty for custom services,
otherwise an error is raised in `dynamic_dns_updater.sh`.

(cherry picked from commit 85ff9db2bffd3d2bf119300d9297f1b524608e1b)
commit fc4c67ed (2022-10-12): Merge pull request #6017 from jcassette/issue-4926-22.03

[22.03] luci-app-ddns: fix creation of custom service (fix #4926)
closed 2022-10-11 by jow-