openwrt/luci #4934: luci_git package is missing

achim71, 2021-03-19:
Seems luci_git ... ipkg got lost.
I tried to install luci on an 19.07.7 build (made with imagebuilder for Netgare R7800) via opkg and got an package luci not found response.
I'd expect the luci_git package here
https://downloads.openwrt.org/releases/19.07.0/packages/arm_cortex-a15_neon-vfpv4/luci/

Fixed it temporary by using an older luci_git package from 19.07.3 from my local download cache.

jow-, 2021-03-19:
Duplicate of #4927 - this is a buildbot issue on the OpenWrt side. I backported buildroot fixes that should address it, but it'll take a few days for the changes to propagate.

closed 2021-03-19 by jow-

achim71, 2021-03-20:
Thank you for the background information.
The latest build attemt this morning also failed. https://downloads.openwrt.org/releases/faillogs-19.07/arm_cortex-a15_neon-vfpv4/luci/luci/compile.txt