openwrt/luci #4935: Missing support for displaying physical port status on the web interface on DSA devices

ygleg, 2021-03-19:
Before the DSA switchover, the link status of physical ports is displayed on the switch page, which is not present on DSA devices anymore. Currently, as far as I can tell, there is no way to acquire this information on LuCI anymore. 
[image]

Now that #4307 has been merged, please consider making these basic info available on the web interface as well. The carrier status, duplex mode and speed can be simply acquired from sysfs:
```
/sys/class/net/eth0/carrier
/sys/class/net/eth0/duplex
/sys/class/net/eth0/speed
```

jow-, 2021-03-19:
Question is where exactly.

I see two potential places:

a) add a new section to the main status page which simply displays all DSA ports with their related link and speed state (basically the just the "Port status: " row)
b) display this info in the bridge-vlan configuration matrix

ygleg, 2021-03-19:
imo option A is preferable, as this would be something that I'd be interested in seeing in a prominent place, either under the "Network" section, or at the very bottom, below the DHCPv6 leases. I don't have anything against option B either other than it might further crowd this menu and make it problematic to display on a narrower screen (mobile) / browser window.

dengqf6, 2021-03-20:
> a) add a new section to the main status page which simply displays all DSA ports with their related link and speed state (basically just the "Port status: " row)

This should be applied to all ethernet ports, not just DSA. Would be useful for routers that have multiple NICs.

camelcamro, 2022-04-04:
nothing done since 1 year ?
any news ?

petiepooo, 2022-05-12:
This is also a feature that I miss from the prior release.  Perhaps it can make it into 22.03?

tricksol, 2022-07-27:
What is the status of this request?

jow-, 2022-07-27:
It is open.

RomanMaronov, 2023-03-12:
Look in https://github.com/openwrt/luci/pull/5918
and https://forum.openwrt.org/t/ethernet-port-icons-on-openwrt/130799/6

adelton, 2025-01-11:
With https://github.com/openwrt/luci/pull/5918 merged, is anything else needed for this issue?