openwrt/luci #4938: Problem. In firewoll

fiduchi, 2021-03-23:
Problem. In firewoll, when creating a parental control rule for mak time limits, when you check the box and click the save and apply button, the checkbox is removed.

jow-, 2021-03-23:
Which "box" do you mean exactly? Can you provide a screenshot?

fiduchi, 2021-03-23:
[image]

fiduchi, 2021-03-23:
This only happens when time limits are enabled.

gearhead, 2021-03-26:
What goes on here is that Luci cannot turn these time based firewall rules on or off. They are always turned off when any change is made. It is easy to check. Make a couple time dependent rules then try to turn them on. Hit save and apply and they are all cleared (turned off). Luci does not seem to be able to change the '0' to '1' for Enabled for these rules. The workaround is to log in via cli and change the 0 to 1 then reload the firewall rules on the command line. It would be awesome if this worked, though.

jow-, 2021-03-29:
@fiduchi and @gearhead - thanks for providing the details, it helped me to track down the cause.

commit 95b5c6cd (2021-03-29): luci-base: ui.js: fix UICheckbox widget operation when tooltips are set

When a tooltip is rendered for a checkbox widget, an additional node is
placed after the checkbox label element, breaking DOM selectors in
bind(), isChecked(), setValue().

Apparently the functionality was never actually tested.

Fixes: #4938
Fixes: e951236e3 ("luci-base: add tooltip handling")
commit adcc2ef4 (2021-03-29): luci-base: ui.js: fix UICheckbox widget operation when tooltips are set

When a tooltip is rendered for a checkbox widget, an additional node is
placed after the checkbox label element, breaking DOM selectors in
bind(), isChecked(), setValue().

Apparently the functionality was never actually tested.

Fixes: #4938
Fixes: e951236e3 ("luci-base: add tooltip handling")

(cherry picked from commit 95b5c6cd6464d11d6baa22bcf2c9469847353813)
closed 2021-03-29 by jow-