openwrt/luci #4945: luci-theme-material input-textarea not displaying correctly.

lsoltero, 2021-03-26:
Initial textarea size seems to be fixed and does not extend to the end of the table cell as it does with the bootstrap and openwrt themes.  Also... the rows as in 

o = s:option(TextValue, "description", translate("Profile Description"))
o.rows = 5  
o.cfgvalue = function(self,section)
    return TextValue.cfgvalue(self,section) or translate("Enter something meaningful here")
end

is ignored.  The size of the initial widget is always the same.

below are images from the material and openwrt themes showing the same field in its intial default state.

[image]

[image]

adelton, 2025-01-11:
What page / application is this? Looking at the `Network > Disagnostics` textarea in OpenWrt 24.10.0-rc5 (r28304-6dacba30a7), it seems to behave reasonably.

systemcrash, 2025-01-11:
Custom. But it's Lua based which is no longer supported.

closed 2025-01-11 by systemcrash