openwrt/luci #4947: Firewall4

oskarirauta, 2021-03-28:
Now that openwrt:master has option of new firewall: firewall4, luci-app-firewall should depend on either firewall or firewall4.

hnyman, 2021-04-01:
@jow- 
Any ideas about the best solution here?

Should firewall4 provide "firewall"?
And should the old firewall package be similarly named to firewall3 and just provide "firewall" ?

I tried compiling the build with firewall4 and noticed that although I disabled firewall3 from the main config, LuCI still pulled it in. Getting it out of the build required editing luci-app-firewall.

jow-, 2021-04-01:
Yes, it should be solved via PROVIDES eventually.

oskarirauta, 2021-04-02:
I have modified my luci-app-firewall to support fw4, first I though se should variate fw3/fw4- but it would be better idea to variate luci-app-firewall due to path inconsistencies.

Lähetetty iPhonesta

oskarirauta, 2021-04-03:
fw4 has a fw3 symbolic link, but if I remember correctly, it’s located in /sbin/fw3 in firewall package - when on firewall4, fw4 binary is in /usr/sbin, and so is symbolic link fw3, so even though applications are compatible between each other, changes have to be made to either, firewall4 or luci-app-firewall or by making variant luci-app-firewall4 provididng luci-app-firewall.

There’s also a helper file which luci-app-firewall depends on, which assists when simpler firewalling rules are made for applications more or less simpler than creating rules, I’ve never before this even known about this, but just found out about this when studying why luci-app-firewall didn’t show in luci; these files are also in different locations, even though it’s paths last directory difference x/x/firewall <—> x/x/firewall4. Pseudos here are since I am bit busy right now and won’t be doing true comparison at the moment..

AND if I remember correctly, there was also a 3rd similar difference. Anyway, my luci-app-firewall at the moment supports fw4 - and changes are not made on running system, changes are on package’s sources so that I can simple build package in my buildroot, if we agree that variant should be on luci’s side - I can easily create a branch for fw4 variant and do a PR - even better would be to support both on a single package, but I wasn’t able to make a depends rules that would had actually worked, although reason for this might had been elsewhere. That kind of package although might be impossible to archieve, since once of these changes were in a rpcd rule file, which cannot be dynamic afaik. 

So let me know your opinions where this variation should be. Other way ofcourse would be to change fw3 link’s location in fw4, but I’d rather loose that link there completely separating these packages completely from each other. Helper file’s location could be worked out on a lua file making it support both locations.

oskarirauta, 2021-04-16:
Okay. It seems to be quite impossible to build a luci package with variant option while using luci.mk and I am quite sure that it isn't acceptable that I tear down a luci package and build it like a conventional package.

I also tried to make a variant in another file, that unfortunately won't work either.
Why not make luci-app-firewall support both, fw3 and fw4? That's quite difficult as well, as both are depending on rcpd acl and menu, where items showing up in interface depend on existence of binary, which is in fw3: /sbin/fw3 and in fw4: /usr/sbin/fw4, later also makes a symlink /usr/sbin/fw3 -> fw4 for compatibility, but this won't help as path must also remain same. rpcd and menu file do not support OR.

Another changes have to be made inside function of file in lici-base that retrieves firewall's helpers. Without this change, page might open but troubles ahead.

Also there seems to be error in code in that file that I will later on open a new ticket.

Hopefully someone will take a look at this idea. Currently I have my own build where luci-app-firewall and luci-app-firewall4 are separate packages, luci and luci-nginx are made to no longer depend on luci-app-firewall, so one needs to choose between them and install either one, but this is not a solution to production version....

closed 2022-03-08 by oskarirauta