openwrt/luci #4948: luci-mod-network: LAN interface setting

castiel652, 2021-03-28:
With latest commit 9ff45c938faf148e99a5c683fe89615a46313769
After at least changing the settings of LAN interface once, LuCI always does this `uci del network.lan.type` to `/etc/config/network` if you click save on the LAN interface setting page.

jow-, 2021-03-28:
Please provide your complete `/etc/config/network` and steps to reproduce it. I am unable to trigger that issue in an x86/64 Qemu image.

castiel652, 2021-03-29:
My apology.

I did more testing and it's because of IGMP snooping

R7800 swconfig kernel 5.4 master branch
Fresh install and before enabling IGMP snooping
```
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdae:1a50:bdda::/48'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth1.1'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option ifname 'eth0.2'
        option proto 'dhcp'

config interface 'wan6'
        option ifname 'eth0.2'
        option proto 'dhcpv6'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '1 2 3 4 6t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '5 0t'
```

After enabling IGMP snooping
```
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdae:1a50:bdda::/48'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth1.1'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option ifname 'eth0.2'
        option proto 'dhcp'

config interface 'wan6'
        option ifname 'eth0.2'
        option proto 'dhcpv6'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '1 2 3 4 6t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '5 0t'

config device
        option name 'br-lan'
        option igmp_snooping '1'
        option multicast_querier '0'
```
After IGMP snooping enabled, changing LAN interface settings will have this behavior
[image]
I won't be able to reach the router if I click Save&Apply

jow-, 2021-03-29:
Thanks @castiel652 - that was helpful.

closed 2021-03-29 by jow-

lcsaszar, 2021-05-25:
I have the same issue. After upgrading luci-mod-network to git-21.140.43486-398982f, it says:

> The existing network configuration needs to be changed for LuCi to function properly. Upon pressing “Continue”, bridges configuration will be moved from “interface” sections to “device” sections the network will be restarted to apply the updated configuration.

After the change my bridged interfaces are no accessible. I can restore fuctionality if I add `option type 'bridge'` back to `config interface 'lan'` section, and add my bridged ports lan1, lan2, lan3, and lan4 with `brctl addif br-lan lan1` command (and same for the other ports). But this is not compatible with LuCi network configuration page.

commit a9a223b9 (2021-03-29): luci-mod-network: don't implicitly move bridge opts into device sections

When setting bridge and device specific options such 'stp' or
'igmp_snooping', LuCI so far transparently created or reused a
`config device` section and set the corresponding option there.

In the case of bridges, this triggers multiple problems:

 - When implicitely creating a `config device` section referring to the
   bridge device, the legacy bridge configuration of the corresponding
   interface is disabled, causing a broken configuration on subsequent
   save operations

 - Netifd does not appear to properly merge  [... cut]
commit 216ddcce (2021-05-27): luci-mod-network: don't implicitly move bridge opts into device sections

When setting bridge and device specific options such 'stp' or
'igmp_snooping', LuCI so far transparently created or reused a
`config device` section and set the corresponding option there.

In the case of bridges, this triggers multiple problems:

 - When implicitely creating a `config device` section referring to the
   bridge device, the legacy bridge configuration of the corresponding
   interface is disabled, causing a broken configuration on subsequent
   save operations

 - Netifd does not appear to properly merge  [... cut]