openwrt/luci #4963: realtime download upload rate wrong calculation

langit7, 2021-04-05:
download rate shown some strange value  ie. huge number download rate  or negative  download/upload rate 
IP Address	Download Rate	Bytes Total	Packets Total
192.168.7.50	12425.51 MB/s	39888.12 MB	27837765 Pkts.
192.168.7.54	2465.88 MB/s	10696.8 MB	5531235 Pkts.
192.168.7.157	1468.62 MB/s	4445.73 MB	3375948 Pkts.
192.168.7.60	1357.5 MB/s	4650.45 MB	4161270 Pkts.
192.168.7.7	1135.37 MB/s	9980.27 MB	7578804 Pkts.
192.168.7.215	808.59 MB/s	2793.1 MB	2304358 Pkts.
192.168.7.50	-41535.2 kB/s	1625.2 MB	16174987 Pkts.
192.168.7.121	-48941.2 kB/s	142.08 MB	964379 Pkts.
192.168.7.50	-55306.85 kB/s	1463.15 MB	15133262 Pkts.
192.168.7.50	-75812.89 kB/s	829.07 MB	7020537 Pkts.

xiaomi mi router 3 pro
OpenWrt 21.02-SNAPSHOT r15927-aacf378608

dragetd, 2021-04-11:
Did you fully load your interface when this happened? Does it happen all the time or just after a while? Did you notice something like this with the current stable version? Something that helps recreating the issue?

langit7, 2021-04-11:
> Did you fully load your interface when this happened? Does it happen all the time or just after a while? Did you notice something like this with the current stable version? Something that helps recreating the issue?

It's happens all the time, I thinks  its because Openwrt-21.02 , I have no issue with v19.07

jlawrence0710, 2022-01-30:
I can confirm the same happens to me and I’m on 21.02 as well.  It’s the only version I’ve used yet.
1948.91 MB/s 11036.47 MB 7922974 Pkts.
61.78 MB/s 348.46 MB 293203 Pkts. 
50.7 MB/s 584.61 MB 504380 Pkts.
23.15 MB/s 94.37 MB 84757 Pkts.

every one of those are completely wrong and impossible as my current network speed is only 5mbps.