openwrt/luci #4964: wireless band select bug in N mode

CWZ7605, 2021-04-05:
With the dual-band wireless module, the 2.4G band cannot be selected in N mode
[image]
[image]

dragetd, 2021-04-11:
The 'Operating frequency' is for compatibility within this band, not for switching between 2,4ghz/5ghz. Every available device should have its own 'radioX' entry, one for 2,4ghz, one for 5ghz and more, if you have multiple devices.

You have radio0 and radio1 and you can configure wireless interfaces in any more on one of these.

It would be helpful if you could rephrase and explain what the issue is. What you expect to happen and what happens instead.
Also, please add which device you are using and which version of OpenWRT, as not every device is equally well supported.

jow-, 2021-04-12:
Not a bug.

closed 2021-04-12 by jow-