openwrt/luci #4970: force upgrade markup (sysupgrade/luci-21.02)

iuncuim, 2021-04-09:
The option "Force upgrade" is outside the container. Tested Firefox/Chrome browsers with reset caches.
[image]
LuCI openwrt-21.02 branch (git-21.088.35541-d627abe) / OpenWrt 21.02-SNAPSHOT r15952+1-6090337

iuncuim, 2021-04-14:
@dibdot the reason is this [commit](https://github.com/openwrt/luci/commit/6ffd1592ce574afe5493d8d7d11f65bec1916e86)

dibdot, 2021-04-14:
I doubt that, rather I wonder why this label was defined as a button!? E.g.
```
[...]
body.push(E('p', {}, E('label', { 'class': 'btn alert-message danger' }, [
[...]
```
remove the "btn" part should fix this IMHO.

iuncuim, 2021-04-14:
> remove the "btn" part should fix this IMHO.
Don't know, maybe this will break other themes.

I made this new rule for bootstrap-theme, now the alert looks ok:
`
.modal .btn.danger {
	white-space: normal;
}
`

closed 2021-04-16 by dibdot

iuncuim, 2021-09-29:
@jow- now other buttons are affected by the problem:
[image]
With new css rule on cascade.css it looks like:
`.modal .btn.danger,
.modal .btn {
	white-space: normal;
}`
[image]
Probably worth reopening the issue.

iuncuim, 2021-09-29:
Patch missed on the 21.02 branch.
https://github.com/openwrt/luci/blob/b39d9bf4bb88acd3120098c3a087e47331d1d757/modules/luci-mod-system/htdocs/luci-static/resources/view/system/flash.js#L280

PR #4988 (2021-04-16): luci-mod-system: remove bogus btn class in flash.js
* fixes #4970
commit 8c6a29a4 (2021-04-16): luci-mod-system: remove bogus btn class in flash.js

* fixes #4970
* sync translations