openwrt/luci #4972: luci-theme-openwrt-2020: does not support table view

feckert, 2021-04-09:
I fell maybe over the following css problem while testing the next stable branches. It looks like the table css definition is missing with **this** LuCI theme. I noticed thos first during my test for luci-app-dockerman with this theme. The luci-app-command also looks strange with this theme. Can anyone confirm this? @jow- do you have a fix for this?

luci-app-commands (dashboard):
[image]

luci-app-commands (config):
The button thing can be fixed by adding btn to the css class.
[image]

hnyman, 2021-04-09:
what is "this" in the "this theme"  ?

feckert, 2021-04-09:
@hnyman no problem :-)

If I add the following to luci-theme-openwrt-2020 cascade.css:

```
diff --git a/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css b/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css
index aa811ef6d..b93fd2715 100644
--- a/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css
+++ b/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css
@@ -270,6 +270,28 @@ body.modal-overlay-active #modal_overlay {
  * table layout
  */

+.table {
+       display: flex;
+       flex-direction: column;
+       width: 100%;
+}
+
+.tr {
+       display: flex;
+       flex-direction: row;
+       flex-wrap: wrap;
+       align-items: flex-end;
+}
+
+.th, .td {
+       flex: 2 2 25%;
+       align-self: flex-start;
+       overflow: hidden;
+       text-overflow: ellipsis;
+       word-wrap: break-word;
+       display: inline-block;
+}
+
 table {
        width: 100%;
        margin: 0 0 1rem 0;
```

Then things look better. But I am not a CSS expert! Maybe there is someone who knows about it ;-)

[image]

jow-, 2021-04-09:
Fixed with these commits:

 - f117b7f1e ("luci-app-commands: use <button> elements")
 - c5ff3244e ("luci-theme-openwrt-2020: honor CSS definitions passed by views")
 - 215586817 ("luci-compat: update markup for CBI TableSections")

closed 2021-04-09 by jow-