openwrt/luci #4991: luci-app-advanced-reboot: Reboot not setting correct ubi (rootfs) partition

arrmo, 2021-04-14:
Hi,

This was working before - I had confirmed it [here](https://forum.openwrt.org/t/ea7300-flash-to-wrong-partition/85238/71?u=arrmo). But now it seems not to be - could be me being fooled, I admit that. But what I'm seeing is that the boot partition is being changed, correct partition being selected for the kernel. But the needed "ubi" partition name is not being set, so it's always booting (rootfs) to the first partition (confirmed in dmesg). A few items I checked,

1) Device info, `ubus call luci.advanced_reboot obtain_device_info`
```
{
        "current_partition": 2,
        "device_name": "Linksys EA7300v2",
        "partitions": [
                {
                        "state": "Alternative",
                        "os": "OpenWrt (Linux 5.10.28)",
                        "number": 1
                },
                {
                        "state": "Current",
                        "os": "OpenWrt SNAPSHOT r16515+6 (Linux 5.4.110)",
                        "number": 2
                }
        ],
        "rom_board_name": "linksys,ea7300-v2"
}
```
2) So the above is correct, and seems to match to, `cat /usr/share/advanced-reboot/devices/linksys-ea7300v2.json`
```
{
        "vendorName": "Linksys",
        "deviceName": "EA7300v2",
        "boardNames": [ "linksys,ea7300-v2" ],
        "partition1MTD": "mtd5",
        "partition2MTD": "mtd7",
        "labelOffset": 32,
        "bootEnv1": "boot_part",
        "bootEnv1Partition1Value": 1,
        "bootEnv1Partition2Value": 2,
        "bootEnv2": "bootcmd",
        "bootEnv2Partition1Value": "run nandboot",
        "bootEnv2Partition2Value": "run altnandboot"
}
```
3) But, mtd naming is not right (for partition 2), `cat /proc/mtd`
```
dev:    size   erasesize  name
mtd0: 00080000 00020000 "boot"
mtd1: 00040000 00020000 "u_env"
mtd2: 00040000 00020000 "factory"
mtd3: 00040000 00020000 "s_env"
mtd4: 00040000 00020000 "devinfo"
mtd5: 00400000 00020000 "kernel1"
mtd6: 02400000 00020000 "ubi"
mtd7: 00400000 00020000 "kernel2"
mtd8: 02400000 00020000 "rootfs2"
mtd9: 00100000 00020000 "sysdiag"
mtd10: 02d00000 00020000 "syscfg"
```

Thoughts? mtd6 is what is also in dmesg, but on reboot it should be mtd8, right? I admit, not sure how the naming is being set - I can't find that part in the code (again, likely me).

Thanks!

stangri, 2021-04-15:
> But the needed "ubi" partition name is not being set, so it's always booting (rootfs) to the first partition (confirmed in dmesg). 

How do you set the needed "ubi" partition name?

hnyman, 2021-04-15:
To my knowledge, the "ubi" name is just assigned for the active rootfs1 or rootfs2 on the fly during the boot process. Note that there is no real partition table or volume labels stored anywhere.

This is from WRT3200ACM, where I first toggled boot manually, and then the second time with advanced_reboot app. Seems to work normally, I think.

```
 OpenWrt SNAPSHOT, r16520-cb3fb45ed1
 -----------------------------------------------------
root@router3:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00200000 00020000 "u-boot"
mtd1: 00020000 00020000 "u_env"
mtd2: 00040000 00020000 "s_env"
mtd3: 00040000 00020000 "devinfo"
mtd4: 001e0000 00020000 "sysdiag"
mtd5: 05000000 00020000 "kernel1"
mtd6: 04a00000 00020000 "ubi"
mtd7: 05000000 00020000 "kernel2"
mtd8: 04a00000 00020000 "rootfs2"
mtd9: 05600000 00020000 "syscfg"
mtd10: 005c0000 00020000 "unused_area"

root@router3:~#

{
        "current_partition": 1,
        "device_name": "Linksys WRT3200ACM",
        "partitions": [
                {
                        "state": "Current",
                        "os": "OpenWrt SNAPSHOT r16520 (Linux 5.4.110)",
                        "number": 1
                },
                {
                        "state": "Alternative",
                        "os": "OpenWrt SNAPSHOT r16416 (Linux 5.4.109)",
                        "number": 2
                }
        ],
        "rom_board_name": "linksys,wrt3200acm"
}
```

SWITCH BOOT AND REBOOT

```
 OpenWrt SNAPSHOT, r16416-180c463526
 -----------------------------------------------------
root@router3:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00200000 00020000 "u-boot"
mtd1: 00020000 00020000 "u_env"
mtd2: 00040000 00020000 "s_env"
mtd3: 00040000 00020000 "devinfo"
mtd4: 001e0000 00020000 "sysdiag"
mtd5: 05000000 00020000 "kernel1"
mtd6: 04a00000 00020000 "rootfs1"
mtd7: 05000000 00020000 "kernel2"
mtd8: 04a00000 00020000 "ubi"
mtd9: 05600000 00020000 "syscfg"
mtd10: 005c0000 00020000 "unused_area"

root@router3:~# ubus call luci.advanced_reboot obtain_device_info
{
        "current_partition": 2,
        "device_name": "Linksys WRT3200ACM",
        "partitions": [
                {
                        "state": "Alternative",
                        "os": "OpenWrt SNAPSHOT r16520 (Linux 5.4.110)",
                        "number": 1
                },
                {
                        "state": "Current",
                        "os": "OpenWrt SNAPSHOT r16416 (Linux 5.4.109)",
                        "number": 2
                }
        ],
        "rom_board_name": "linksys,wrt3200acm"
}
```

Use advanced_reboot to toggle...

```
 OpenWrt SNAPSHOT, r16520-cb3fb45ed1
 -----------------------------------------------------
root@router3:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00200000 00020000 "u-boot"
mtd1: 00020000 00020000 "u_env"
mtd2: 00040000 00020000 "s_env"
mtd3: 00040000 00020000 "devinfo"
mtd4: 001e0000 00020000 "sysdiag"
mtd5: 05000000 00020000 "kernel1"
mtd6: 04a00000 00020000 "ubi"
mtd7: 05000000 00020000 "kernel2"
mtd8: 04a00000 00020000 "rootfs2"
mtd9: 05600000 00020000 "syscfg"
mtd10: 005c0000 00020000 "unused_area"
root@router3:~# ubus call luci.advanced_reboot obtain_device_info
{
        "current_partition": 1,
        "device_name": "Linksys WRT3200ACM",
        "partitions": [
                {
                        "state": "Current",
                        "os": "OpenWrt SNAPSHOT r16520 (Linux 5.4.110)",
                        "number": 1
                },
                {
                        "state": "Alternative",
                        "os": "OpenWrt SNAPSHOT r16416 (Linux 5.4.109)",
                        "number": 2
                }
        ],
        "rom_board_name": "linksys,wrt3200acm"
}
```

arrmo, 2021-04-15:
This does help, thanks! It seems that the issue is the wrong rootfs being selected. I need to keep digging ... it was working fine a while back, but now seems to be broken (again?). Not necessarily in luci-app-advanced-reboot, just generically.

Thanks again.

arrmo, 2021-04-16:
OK, will close this out - figured out the (my 😆) issue. If you toggle => don't try to use `obtain_device_info` then (without a reboot) ... it won't work right. I can explain if you want, but from the code I can see what is going on.

Thanks!

closed 2021-04-16 by arrmo