openwrt/luci #4992: Home page display error

643406834, 2021-04-15:
[image]

hnyman, 2021-04-15:
I am not OpenWrt with x86 expert, but I think that the x86 builds are not device-specific, and so no device name (like WRT3200ACM) is encoded in the image. So, I am not sure that there is any bug at all.

dibdot, 2021-04-15:
That's not a LuCI issue at all, we're just parsing /etc/board.json.

dibdot, 2021-05-02:
No more feedback by TO - closing this for now.

closed 2021-05-02 by dibdot

stangri, 2021-05-03:
BTW, If the model is correctly identified, it is correctly shown in WebUI (at least on 21.02.0-rc1):
[image]
This is definitely not an apropriate repo for this issue.