openwrt/luci #4993: DynamicList does not show in the GridSection (luci-20.02, js)

iuncuim, 2021-04-16:
I made a simple page that used DynamicList and GridSection.
In edit mode, the data is displayed normally, but it is not displayed in tab form.
If I use an "option" in the configuration file instead of "list", it is displayed correct.
[image]

Config file in this case:
```
config pingcontrol 'pingcontrol'
	option check_period '30'
	option sw_before_modres '3'
	option sw_before_sysres '0'
	option enabled '0'
	option iface 'lan'
	list testip '1.1.1.1'

config pingcontrol 'dddddd'
	option iface 'lan'
	option check_period '60'
	option sw_before_modres '77'
	option sw_before_sysres '0'
	option testip '192.168.1.11 1.1.1.1 2.2.2.2 3.3.3.3 4.4.4.4'
	option enabled '1'
```
Web page:
```
		m = new form.Map('pingcontrol', _('PingControl'));
		m.description = _("Server availability check");

		s = m.section(form.GridSection, 'pingcontrol', _('Settings'));
		s.addremove = true
		s.nodescriptions = true

		o = s.option(form.DynamicList, 'testip', _('IP address of remote servers'));
		o.rmempty = false;
		o.datatype = 'ipaddr';
```
@hnyman I found an old topic about [lua](https://forum.openwrt.org/t/how-do-i-add-list-in-a-config-file-instead-of-option-in-luci-using-cbi/49913). It is now possible for luci (with js frontend) to generate option instead of list?

jow-, 2021-04-16:
@iuncuim - the underlying bug in form.js is fixed. If you can't wait for the fix (or if you'd like to present the values differently anyway) you could override `textvalue()`, e.g.:

```javascript
o = s.option(form.DynamicList, 'testip', _('IP address of remote servers'));
o.rmempty = false;
o.datatype = 'ipaddr';
o.textvalue = function(section_id) {
	var cval = L.toArray(this.cfgvalue(section_id));
	return cval.length ? '%h'.format(cval.join(', ')) : E('em', _('No IP addresses configured'));
};

```

closed 2021-04-16 by jow-

iuncuim, 2021-04-16:
@jow- Thank for your fix and example. Nice work!

iuncuim, 2021-05-06:
@jow- Could you explain this is bug or my mistake?
When I add this construction to gridsection:
```
var callFileList = rpc.declare({
	object: 'file',
	method: 'list',
	params: [ 'path' ],
	expect: { entries: [] },
	filter: function(list, params) {
		var rv = [];
		for (var i = 0; i < list.length; i++)
			if (list[i].name.match(/^tty[A-Z]/) || list[i].name.match(/^com[0-9]/))
				rv.push(params.path + list[i].name);
		return rv.sort();
	}
});
...
		o = s.option(form.Value, 'devicename', _('Port'));
		o.rmempty = false;
		o.load = function(section_id) {
			return callFileList('/dev/')
			.then(L.bind(function(devices) {
				for (var i = 0; i < devices.length; i++)
					this.value(devices[i]);
				this.value(i);
				return form.Value.prototype.load.apply(this, [section_id]);
			}, this));
		};
```
I get this: 
[image]
With other type of section I get normal list of ports without iterance

commit 5c792aef (2021-04-16): luci-base: form.js: fix AbstractValue.textvalue() for uci list options

Serialize the uci list value into a space separated string before passing
it to String.format() for HTML escaping. Without that change, empty strings
were returned whenever the underlying uci get operation yieled an array.

Fixes: #4993
commit 06dd6b52 (2021-04-16): luci-base: form.js: fix AbstractValue.textvalue() for uci list options

Serialize the uci list value into a space separated string before passing
it to String.format() for HTML escaping. Without that change, empty strings
were returned whenever the underlying uci get operation yieled an array.

Fixes: #4993

(cherry picked from commit 5c792aefc744d1417fc9e24cfb92cd61cf8a651f)
commit d763f976 (2021-04-16): luci-base: form.js: fix AbstractValue.textvalue() for uci list options

Serialize the uci list value into a space separated string before passing
it to String.format() for HTML escaping. Without that change, empty strings
were returned whenever the underlying uci get operation yieled an array.

Fixes: #4993

(cherry picked from commit 5c792aefc744d1417fc9e24cfb92cd61cf8a651f)