openwrt/luci #5004: luci-mod-network: Interface wrongly re-assigned when port VLAN membership changed

farfade, 2021-04-21:
Hi,

I'm not sure why the openwrt UI decides to auto-reassign interface mappings when I change LAN 4 on VLAN 3 from tagged to untagged ?

[image]

Indeed, I am forced to get back to the interfaces to remap eth0.3 to GUEST before applying in order to make things work.

Or is something I misunderstand in my configuration ?

Cheers,

farfade

_Originally posted by @farfade in https://github.com/openwrt/luci/issues/3709#issuecomment-823457460_

jow-, 2021-04-21:
Please provide the contents of `/etc/board.json` as well as the complete `/etc/config/network´.

farfade, 2021-04-21:
Thank you jow for your interest.

Please find here the files required.

[etc_board_json.txt](https://github.com/openwrt/luci/files/6351093/etc_board_json.txt)
[etc_config_network.txt](https://github.com/openwrt/luci/files/6351094/etc_config_network.txt)