openwrt/luci #5006: I can't select dockerman

NickSJ08, 2021-04-21:
Is it the kernel version question?
/usr/lib/lua/luci/cbi.lua:53: /usr/lib/lua/luci/model/cbi/dockerman/overview.lua:90: unexpected symbol near '<'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/cbi.lua:53: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1353: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

hnyman, 2021-04-21:
Sounds like a tag formatting issue.
Which OpenWrt version you are using?
and which theme?

NickSJ08, 2021-04-21:
Which OpenWrt version you are using?  
OpenWrt 19.07 kernel version 5.4.113   FW OpenWrt SNAPSHOT r16577-b832ebb886 / LuCI Master git-21.109.51834-c05bd19
and which theme? default bootstrap

hnyman, 2021-04-21:
Dockerman app only exists for master and 21.02.
It can't be used with 19.07

closed 2021-04-21 by hnyman

NickSJ08, 2021-04-21:
I don't know, try again tomorrow
thank you for your reply

feckert, 2021-04-21:
I have found the error!
https://github.com/openwrt/luci/commit/ad05a0187bb9b03ba5ac9df27159f54521e15e18#diff-8b6af37433f463b35884d7b94afbd4fd03d5c9872efb1bbb040a46f20db7525bR90
Something must have gone wrong during rebasing!!!!
I can't explain it to myself!
Thanks for the tip I will fix it immediately

feckert, 2021-04-21:
Fixed with commit https://github.com/openwrt/luci/commit/c4f3ac05288ba148fcd5c0222f34f393dbec57a2 in master
Fixed with cherry-pick commit https://github.com/openwrt/luci/commit/a7845594f5b93cb7dd6abb201db66e4348126fc8 in openwrt-21.02

@NickSJ08  thank you for pointing it out.

NickSJ08, 2021-04-22:
@feckert Can it be used on 19.07 after modification?

feckert, 2021-04-22:
Docker does not exist for the branch openwrt-19.07

If you do it in your own build to port dockerd on the stable branch openwrt-19.07 then there should be nothing to stop the luci-app-dockerman running on that branch.

**But I would like to point out that this is not supported from upstream.**
**And you will not get any support.**
**This is then at your own risk.**

NickSJ08, 2021-04-23:
I use openwrt-21.02 , but still show error page (Docker->Overview).

/usr/lib/lua/luci/cbi.lua:53: /usr/lib/lua/luci/model/cbi/dockerman/overview.lua:90: unexpected symbol near '<'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/cbi.lua:53: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1353: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

$git status
On branch openwrt-21.02
Your branch is up to date with 'origin/openwrt-21.02'.

nothing to commit, working tree clean
$git pull
Already up to date.

Linux OpenWrt 5.4.111 #0 SMP Sun Apr 18 10:06:57 2021 x86_64 GNU/Linux

hnyman, 2021-04-23:
Not quite sure what you are doing, but some advice:
* You do not use git in the router
* Luci sources are in the feed in the build system, so there you can update sources and then compile Luci
* Even in the build system, the usual way is via feeds update command (unless you have navigated to the feed repo)
* You would need to compile the dockerman package first. Git has just the sources, not the compiled app

* in router, you can just install the updated package via opkg (assuming that the buildbot has already compiled fixed Luci for your architecture)
  opkg update
  opkg upgrade luci-app-dockerman

Or download the .ipk from the download repo and use opkg to install it

NickSJ08, 2021-04-23:
I currently use openwrt-21.02 and install luci-app-dockerman
But the Docker->Overview page of the luci dockerman interface cannot be displayed
The page is displayed as follows:

/usr/lib/lua/luci/cbi.lua:53: /usr/lib/lua/luci/model/cbi/dockerman/overview.lua:90: unexpected symbol near '<'
stack traceback:
[C]: in function 'assert'
/usr/lib/lua/luci/cbi.lua:53: in function 'load'
/usr/lib/lua/luci/dispatcher.lua:1353: in function '_cbi'
/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

Do I need to modify anything?

feckert, 2021-04-23:
Maybe you need to clear the luci cache or restart the router.
Also, I see that not all builds for all architectures have gone through now with my change
https://buildbot.openwrt.org/master/packages/#/changes/566