openwrt/luci #5009: luci add pptp interface mtu option missing

ptpt52, 2021-04-23:
I cannot find any web ui for option mtu while adding pptp interface

ptpt52, 2021-04-23:
I notice that, in file: protocols/luci-proto-ppp/htdocs/luci-static/resources/protocol/pptp.js
```
                o = s.taboption('advanced', form.Value, 'mtu', _('Override MTU'));
                o.placeholder = dev ? (dev.getMTU() || '1500') : '1500';
                o.datatype    = 'max(9200)';
```
there is code for mtu option, but somehow missing in web ui, why?

feckert, 2021-04-23:
I can confirm this. If I use a name other than mtu, then the option appears.
I'll have to take a closer look to see if this is related to the default options for all protocols.

ptpt52, 2021-04-23:
the same situation is when add interface with proto ppp

ptpt52, 2021-04-23:
it seems mtu option is place on physical tab, but no physical tab for ppp/pptp

rmilecki, 2021-06-10:
"Override MTU" should be present in the Advanced tab. I did quick test using 21.02-rc2:

[image]

[image]

Please provide screenshot from you interface

ptpt52, 2021-06-10:
in the current version, it is fixed.

closed 2021-06-10 by rmilecki