openwrt/luci #5013: Unable to edit lists in LuCI

jernejs, 2021-04-25:
There doesn't seem to be any way to edit a list entry in LuCI - eg. Network → DHCP and DNS → General settings → DNS forwardings - you can only delete and add entries, but you can't edit an existing entry, despite the UI giving appearance of a text entry field:
[image]
The example text is also nearly the same shade as existing list entries, making it really hard to distinguish.

LuCI openwrt-19.07 branch (git-20.136.49537-fb2f363) / OpenWrt 19.07.3 r11063-85e04e9f46

guidosarducci, 2021-04-26:
I think this is one of the bigger usability regressions that came with switching to client-side JS, making list editing difficult and awkward. Not sure if it applies beyond `form.DynamicList` usage however.

@jow- Can this be improved, ideally for 21.02 as well?

jernejs, 2021-04-26:
If it can't be fixed (yet), at least make the non-editable text look sufficiently different from actual textboxes.

adelton, 2025-01-11:
On OpenWrt 24.10.0-rc5 (r28304-6dacba30a7), hovering over the existing (noneditable) field shows a cross cursor:
[image]
And selecting the field with keyboard with Tab does not put a cursor in there, unlike to the last, editable field with plus next to it.

Not sure if any more visual distinction is needed (given there hasn't been any traction in this issue since 2021).

systemcrash, 2025-01-14:
A rendered field is safer (and not susceptible to screen-scraping) from browser auto-fill, and is possibly one reason it was done this way.

muink, 2025-01-14:
@stokito I think it's technically feasible.

[1.webm](https://github.com/user-attachments/assets/53e60d92-96d2-4913-8e65-5c64a934d03f)

Need to modify

``` js
<div class="cbi-value-field">
  <div id="cbid.fchomo.Proxy_host.payload2_type" class="cbi-dynlist" data-idref="3c019a8d" data-changed="true">
    <div class="item" tabindex="0" draggable="true">
      <span>RULE-SET</span> // <--- this value
      <input type="hidden" value="RULE-SET" style="font-family: sans-serif;"> // <--- this value
    </div>
    <div class="item flash" tabindex="0" draggable="true">
      <span>SRC-PORT</span>
      <input type="hidden" value="SRC-PORT" style="font-family: sans-serif;">
    </div>
    <div class="add-item control-group">
      ...
    </div>
  </div>
</div>
```
and generate [cbi-dynlist-change](https://github.com/openwrt/luci/pull/7533) event.