openwrt/luci #5022: luci-app-mwan3: Renaming the current Load balancing label in LuCI navigation

jamesmacwhite, 2021-04-29:
@feckert @aaronjg 

This is something minor but currently the mwan3 navigation menu label is "Load balancing" in LuCI for both the Status and Network menu. Because of the combination of both load balancing and failover functionality, should the label be renamed to something that is more encompassing of it's feature set e.g. "Multi WAN".

**Examples could include:**

For the status menu:

* Multi WAN Status
* mwan3 Status

For the network menu:

* Multi WAN
* mwan3 Interfaces
* Multi WAN Configuration

Having it labelled as load balancing doesn't seem accurate given it's capabilities and uses. I'd be open to other suggests but I think having it labelled as Load Balancing isn't really accurate.

feckert, 2021-04-30:
@jamesmacwhite I have had a patch in my patch set for some time that changes this to `MultiWAN Manager`
```
From: Florian Eckert <fe@dev.tdt.de>
Date: Thu, 6 Apr 2017 12:01:34 +0200
Subject: [PATCH]  luci-app-mwan3: change menu entry name

---
 applications/luci-app-mwan3/luasrc/controller/mwan3.lua | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/applications/luci-app-mwan3/luasrc/controller/mwan3.lua b/applications/luci-app-mwan3/luasrc/controller/mwan3.lua
index 1fb9083a53c1a4ca639c8bb9b0ed711abdad008f..f45351e3bb147afd7467bd6eb1d8e396114ee93e 100644
--- a/applications/luci-app-mwan3/luasrc/controller/mwan3.lua
+++ b/applications/luci-app-mwan3/luasrc/controller/mwan3.lua
@@ -16,7 +16,7 @@ function index()

        entry({"admin", "status", "mwan"},
                alias("admin", "status", "mwan", "overview"),
-               _("Load Balancing"), 600).acl_depends = { "luci-app-mwan3" }
+               _("MultiWAN Manager"), 600).acl_depends = { "luci-app-mwan3" }

        entry({"admin", "status", "mwan", "overview"},
                template("mwan/status_interface"))
@@ -38,7 +38,7 @@ function index()

        entry({"admin", "network", "mwan"},
                alias("admin", "network", "mwan", "interface"),
-               _("Load Balancing"), 600).acl_depends = { "luci-app-mwan3" }
+               _("MultiWAN Manager"), 600).acl_depends = { "luci-app-mwan3" }

        entry({"admin", "network", "mwan", "globals"},
                cbi("mwan/globalsconfig"),
--
```

jamesmacwhite, 2021-04-30:
I think that makes sense. More descriptive and accurate.

jamesmacwhite, 2021-05-01:
@feckert I think it is worth applying your 2017 patch before 21.02 gets a formal release, what do you think?

PR #5174 (2021-07-12): luci-app-mwan3: convert to JS
I have finally found the time to convert the luci-app-mwan3 to java script.

Todo after merge:

- Currently, the plausibility checks are only done for IPv4. But that is not correct see #4603. The interface can now also be IPv6 only. This still needs to be worked on.
- Update I18N

Dependency:

- In order for the troubleshoot output to work again, this [P/R](https://github.com/openwrt/packages/pull/16066) must be merged before. The query is better off there.

Info:

- @jamesmacwhite I have also unified the naming from `Mwan3` to `MultiWAN Manager. fixes #5022
closed 2021-07-12 by feckert