openwrt/luci #5025: Luci theme material text overflow

thecode, 2021-05-01:
[image]

[image]

Issue exists both on OpenWRT 19.0.7 with `luci-theme-material` `git-21.096.73306-254083c-1`
And on OpenWRT 21.02.0-rc1 with `luci-theme-material` `git-20.304.74050-01cefed`

stangri, 2021-05-02:
I'm unable to reproduce on 21.02.0-rc1 even with the page zoom set to 110% in Chrome. What's your device/browser/page zoom?

thecode, 2021-05-02:
Tested under 2 Windows 10 (on two different machines) with the following browsers:
- Google Chrome Version 90.0.4430.93 (Official Build) (64-bit)
- Microsoft Edge Version 90.0.818.51 (Official build) (64-bit)

Tested with 1920x1200 and 1920x1080 monitors, with different display scaling: 100% and 125%
Reproduced on any page zoom between 50% and 250% (e.g. 50, 67, 75, 100, 150, 175)

However it is related to the aspect ratio of the page, see the example bellow when the the page is narrow (how much depends on the device and zoom, it looks ok. Both pictures bellow are using the same zoom (100%) on the same browser (Chrome), but even it is ok on the narrow you can still see the text is on the button side of the border.
Narrow:
[image]
Wide:
[image]

Thanks for looking into it.

jorgebnunes, 2021-05-14:
Hi, I have the same problem.
If you edit /www/luci-static/material/cascade.css, search for 

.cbi-progressbar::after{font-family:monospace;font-size:1.3em;font-weight:700;

and change "1.3em" to "1.0em"

And voilá, problem solved :-)

thecode, 2023-01-22:
Closing this issue, although this is still not fully fixed (text is still 1-2 pixels off),  text size was reduced (last tested with `git-22.339.35089-7af4a64`) so it no longer overlaps.
[image]

closed 2023-01-22 by thecode