openwrt/luci #5026: /usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be currupt, unable to find section "main"

BUG-reporter, 2021-05-01:
After upgrading from 19.07.7 to 21.02.0-rc1 (with previous configuration saving) Luci is not working with the following message while trying to enter it:
[image]

dibdot, 2021-05-02:
Most probably not a LuCI bug, check your config (see https://github.com/openwrt/luci/issues/4758 for reference).

closed 2021-05-02 by dibdot

BUG-reporter, 2021-05-02:
Config is default immediately after upgrading.

dibdot, 2021-05-03:
Did you check/correct the config as mentioned in the referenced ticket - esp. the ubus path?

alan636, 2023-03-23:
I am sure that "/usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be corrupt, unable to find section 'main'" is a bug and continues happening in OpenWrt 21 version. Because after I used the stress test tool named [Jmeter](https://github.com/apache/jmeter) with 2,000 HTTP connection threads of an internet website then the Openwrt main page lose the response and showed that text in the browser when I refresh. 

[image]
[image]

alan636, 2023-03-23:
> I am sure that "/usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be corrupt, unable to find section 'main'" is a bug and continues happening in OpenWrt 21 version. Because after I [...]

It is not ruled out that the low performance of the router is the cause, but even the low performance of the router should ensure the normal operation of the management interface webpage, or restore the normal operation of the management interface webpage after the pressure drops.

jow-, 2023-03-23:
This error also could mean that the `rpcd` service crashed and did not recover, check your routers `logread` for details.