openwrt/luci #5039: Static lease 'instance' option not used

pgaskin, 2021-05-06:
When creating a static lease for a single dnsmasq instance with `option instance`, LuCI still shows it in all instances. In addition, it would be useful to have an option in the GUI to choose whether the lease is instance-specific.

pgaskin, 2021-05-09:
I thought about this a bit, and I think a checkbox would be the most intuitive. I'm going to attempt to implement this, but it will be my first time working with the LuCI code, so it might take a while. I've managed to figure out how to add it as a dropdown or a textbox, but I'm currently trying to figure out how to add a custom checkbox not directly mapped to a UCI setting and how to get CBI to filter the list.

pgaskin, 2021-05-09:
Is there a way to get the current section instance name synchronously from within the render function?

systemcrash, 2025-01-11:
That requires the page knowing state (which happens after render) before render. I've looked at this before and I've found no easy solution for this yet (which seems available within many other pages).