openwrt/luci #5040: Luci theme OpenWrt2020 - Firewall page blank

cybrnook, 2021-05-06:
Using latest 21.02 snapshot, current as of today (05/06/2021) the Firewall page just displays as blank. This was tested on latest edge chromium. I have noticed this on the Master branch as well:

[image]

jow-, 2021-05-07:
Can you press F12 and look for errors in the browser console?

cybrnook, 2021-05-07:
Nothing life changing stood out. I took a screenshot of the main page and the firewall page. Unless I am looking at the wrong place, nothing major stood out to me. As a side note, I tested with latest firefox as well, and it's the same behavior.

Main Page:
[image]

Firewall Page:
[image]

jow-, 2021-05-07:
Seems the issue is triggered when ip6tables is removed from the system.

cybrnook, 2021-05-07:
That would make sense since I have IPV6 disable in my .config.

closed 2021-05-07 by jow-

cybrnook, 2021-05-07:
Thanks @jow-

commit e9b6431b (2021-05-07): luci-mod-status: iptables: always make tab pane visible

When ip6tables is not present, the tab group was not initialized,
causing the iptables status display to be empty with certain themes.

Manually mark the pane active in this case to avoid the problem.

Fixes: #5040

(cherry picked from commit d24e3295f004c87722ad59927738f208a2a84ac5)