openwrt/luci #5044: luci-app-lxc: complaining "no templates for your architecture (x86_64)" while running on a x86_64 device.

hathlife, 2021-05-10:
[image]
somehow the app (or lxc backend, maybe?) sent `x86_64` as name of arch instead of the correct one `amd64` to the container image site, which should not happen.

dibdot, 2021-05-14:
Which container image site/url did you use, e.g. what's configured in /etc/config/lxc?

hathlife, 2021-05-15:
> Which container image site/url did you use, e.g. what's configured in /etc/config/lxc?

Both`images.linuxcontainer.org` and `repo.turris.cz/lxc` share the same error.

and here's `/etc/config/lxc` details:
```
config lxc 'lxc'
        option ssl_enabled '1'
        option min_space '100000'
        option min_temp '100000'
        option url 'repo.turris.cz/lxc'
```

dibdot, 2021-05-15:
The turris repo doesn't provide images for x86 or amd64, so your current config is invalid.

I guess (I don't use lxc anymore), that the download url of linuxcontainers has changed or will be redirected or similiar. The current code maps x86_64 to amd64 only for 'images.linuxcontainers.org', see for reference:

https://github.com/openwrt/luci/blob/3c66c5b1651aa25afbff09bee45047da9a0ba43d/applications/luci-app-lxc/luasrc/controller/lxc.lua#L162

To get this working again, you should get a valid download location and change the function in that way, that the mapping only happens, if it does **not** match the turris repo.

dibdot, 2021-06-12:
No more feedback by TO - closing this.

closed 2021-06-12 by dibdot