openwrt/luci #5067: csstidy break font-face

jjm2473, 2021-05-25:
Say there is a test.css:
```css
@font-face {
  font-family: 'argon';

  src: url('fonts/argon.eot?vncylf#iefix') format('embedded-opentype'), url('fonts/argon.ttf?vncylf') format('truetype'), url('fonts/argon.woff?vncylf') format('woff'), url('fonts/argon.svg?vncylf#argon') format('svg');
  font-weight: normal;
  font-style: normal;
  font-display: block;
}
@font-face {
  font-family: 'TypoGraphica';
  src: url('fonts/TypoGraphica.eot?#iefix') format('embedded-opentype'), url('fonts/TypoGraphica.woff') format('woff'), url('fonts/TypoGraphica.ttf') format('truetype'), url('fonts/TypoGraphica.svg#TypoGraphica') format('svg');
  font-weight: normal;
  font-style: normal;
}
```

run `./staging_dir/hostpkg/bin/csstidy test.css --remove_last_semicolon=true tmp.css`, got tmp.css:
```css
@font-face {
font-display:block;
font-family:TypoGraphica;
src:url(fonts/TypoGraphica.eot?#iefix) format('embedded-opentype'),url(fonts/TypoGraphica.woff) format('woff'),url(fonts/TypoGraphica.ttf) format('truetype'),url(fonts/TypoGraphica.svg#TypoGraphica) format('svg');
font-weight:400;
font-style:normal
}
```

jjm2473, 2021-05-25:
https://stackoverflow.com/questions/32998239/csstidy-breaking-font-face

jjm2473, 2022-10-17:
@jow- Is this bug sitll present?

jow-, 2022-10-17:
Sure, I didn't work on it as I do not maintain the tool. If your theme is affected, consider setting `LUCI_MINIFY_CSS:=0` in its Makefile to opt out of csstidy, regardless of the menuconfig setting.

jjm2473, 2022-10-18:
OK, thanks

closed 2022-10-18 by jjm2473