openwrt/luci #5068: luci-mod-network: bridge configuration breaks existing config

lcsaszar, 2021-05-25:
After upgrading luci-mod-network to git-21.140.43486-398982f, it says:

> The existing network configuration needs to be changed for LuCi to function properly. Upon pressing “Continue”, bridges configuration will be moved from “interface” sections to “device” sections the network will be restarted to apply the updated configuration.

After the change my bridged interfaces are no accessible. I can restore fuctionality if I add `option type 'bridge'` back to `config interface 'lan'` section, and add my bridged ports lan1, lan2, lan3, and lan4 with `brctl addif br-lan lan1` command (and same for the other ports). But this is not compatible with LuCi network configuration page.

jow-, 2021-05-25:
Please attach your original /etc/config/network and the output of `opkg list-installed netifd`

lcsaszar, 2021-05-25:
`config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd31:xxxx:xxxx:xxxx::/64'

config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option ifname 'lan1 lan2 lan3 lan4'

config interface 'wan'
        option ifname 'wan'
        option proto 'dhcp'
        option peerdns '0'
        list dns 'xxxx'
        list dns 'xxxx'

config interface 'wan6'
        option proto '6in4'
        option peeraddr 'xxxx'
        option ip6addr 'xxxx::2/64'
        list ip6prefix 'xxxx::/64'
        option tunnelid 'xxxx'
        option username 'xxxx'
        option password 'xxxx'
        option force_link '1'
        option ip6assign '64'
        option defaultroute '0'
        option peerdns '0'
        list dns 'xxxx'
        list dns 'xxxx'
        option mtu '1480'`

`root@OpenWrt:/# opkg list-installed netifd
netifd - 2021-04-03-327da989-3
`

lcsaszar, 2021-05-25:
After upgrading the kernel to 5.10.39 it works.

closed 2021-05-25 by lcsaszar

ezplanet, 2021-05-28:
Same here, I had to revert to my previous build r16090 because nothing worked.

Are these upgrades actually tested?

Reply to: openwrt/luci ***@***.***>
Date: Tuesday, 25 May 2021 at 10:15
To: openwrt/luci ***@***.***>
Cc: Subscribed ***@***.***>
Subject: [openwrt/luci] luci-mod-network: bridge configuration breaks existing config (#5068)

ezplanet, 2021-05-30:
It looks like we are being left with a now broken and unusable master branch. When is this going to be fixed?

ezplanet, 2021-05-30:
Why has this issue been closed? It isn't resolved!

stangri, 2021-05-31:
@ezplanet if you're using a development branch, there's a chance that things will break. If you want to help get things fixed, do consider posting the information requested [here](https://github.com/openwrt/luci/issues/5068#issuecomment-847716858).

rmilecki, 2021-06-09:
For a reference: @lcsaszar was using old OpenWrt with old `netifd` (2021-04-03) that didn't support the new `network` syntax.

I didn't think originally that people may try recent LuCI with outdated OpenWrt.

That problem was fixed in the commit 17af33ee48bb ("luci-mod-network: migrate network config depending on netifd version")

> After upgrading the kernel to 5.10.39 it works.

Kernel update has nothing to do with it. You switched to OpenWrt with updated `netifd` that also happened to have updated kernel.