openwrt/luci #5069: Wireless: Modifications to wifi-iface or wifi-device can change network.$NET.ifname

pmelange, 2021-05-25:
Tested with 19.07.7

We have come across a problem with Freifunk.  This issue is: if someone wants to change (for example) the channel width of the 5GHz radio from 20MHz to 40MHz, then in the network config the value of network.$NET.ifname gets set to the wifi-iface section name.

To reproduce this, have the following setup in wireless:
* Here the section name is ```wlan0_lan_5``` and the ifname is ```wlan0-lan-5```.
* We use named sections like ```wlan0_mesh_5``` so that we can properly programmatically modify the setting for client/mesh interfaces.
```
config wifi-iface 'wlan0_lan_5'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
	option ifname 'wlan0-lan-5'
```
And in network (standard config):
```
config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
```

After clicking "Save" then viewing "Unsaved Changes" the following is shown:
[image]

"Save & Apply" causes the network interface to be ```network.lan.ifname='eth0.1 wlan0_lan_5'``` which is not correct.  Note, the ifname in the wireless config is wlan0-lan-5 while the section name is wlan0_lan_5.  Also, the wireless section name should simply not be included in the network config.

The real problem comes with 802.11s mesh wifi-ifaces with OLSRd as the routing daemon.  The interface stops working properly.  Before "Save & Apply" the wifi-iface and interface section looks like this:
```
config wifi-iface 'wlan0_mesh_5'
	option network 'wireless0'
	option device 'radio0'
	option ifname 'wlan0-mesh-5'
	option mode 'mesh'
	option encryption 'none'
	option mesh_fwding '0'
	option mesh_id 'Mesh-Freifunk-Berlin'
	option mcast_rate '12000'
	option mesh_rssi_threshold '0'
```
```
config interface 'wireless0'
	option netmask '255.255.255.255'
	option proto 'static'
	option ipaddr '10.31.39.132'
	option ip6assign '64'
```
and ```ip a``` should look like this:
```
26: wlan0-mesh-5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether f8:1a:67:5a:a2:4c brd ff:ff:ff:ff:ff:ff
    inet 10.31.39.132/32 brd 255.255.255.255 scope global wlan0-mesh-5
       valid_lft forever preferred_lft forever
    inet6 fddd:f2e:e74e:11::1/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::fa1a:67ff:fe5a:a24c/64 scope link 
       valid_lft forever preferred_lft forever
```

After "Save & Apply" the interface section changes to this:
```
config interface 'wireless0'
	option netmask '255.255.255.255'
	option proto 'static'
	option ipaddr '10.31.39.132'
	option ip6assign '64'
	option ifname 'wlan0_mesh_5'
```
And as a visual representation:

[image]

Also, the interface no longer gets an IP address assigned
```
24: wlan0-mesh-5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether f8:1a:67:5a:a2:4c brd ff:ff:ff:ff:ff:ff
    inet6 fe80::fa1a:67ff:fe5a:a24c/64 scope link 
       valid_lft forever preferred_lft forever
```

I see this as a critical problem which needs to be solved.

pmelange, 2021-11-10:
@jow- It seems like this issue is still persisting (at least in 21.02.x).  Is there any chance that this will be looked at?

jow-, 2021-11-10:
Can't reproduce it locally. Can you arrange SSH access on an affected device? You can find my SSH public key at http://mein.io/id_rsa.pub

pmelange, 2021-11-10:
Here are the steps to reproduce this in 21.02.1

wireless config before using luci
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'ffe09000.pcie/pci9000:00/9000:00:00.0/9000:01:00.0'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'wlan0_lan_5'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option ifname 'wlan0-lan-5'
```
and network config looks like this
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
```

Change the 5GHz radio from 20MHz to 40MHz

review of the changes are:

[image]

And the router will time-out on "save and apply" and revert the changes.

wireless config changes to this:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'ffe09000.pcie/pci9000:00/9000:00:00.0/9000:01:00.0'
        option disabled '0'
        option htmode 'HT40'
        option cell_density '0'

config wifi-iface 'wlan0_lan_5'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option ifname 'wlan0-lan-5'
```

and more importantly, network config changes to this
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config interface 'lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option device 'br-lan wlan0_lan_5'
```

pmelange, 2021-11-10:
note, if config wifi-iface is set to "wifinet0" then everything works as expected.

when config wifi-iface is set to (for example) "wlan0lan5" or "wlan0_lan_5" the problem shows itself.

jow-, 2021-11-10:
Thanks, the naming was significant. Fixes pushed to all relevant branches.

closed 2021-11-10 by jow-

pmelange, 2021-11-10:
Wonderful!  I also tested the change locally, both with a wireless interface bridged to physical ports and a wireless interface which has no physical ports.  Really great!!!

commit d4092b15 (2021-11-10): luci-base: network.js: ignore wireless ifname patterns on retrieving devices

Only treat the given identifier as Linux netdev name if we can find a
corresponding entry in the device info cache and do not consider strings
starting with "wlan", "ath" or "wl" to be existing devices.

This fixes incorrectly adding wireless sections as ifnames to network
interfaces when the wifi-iface section name begins with one of the
`iface_patterns_wireless` patterns.

Fixes: #5069
commit e1ccb662 (2021-11-10): luci-base: network.js: ignore wireless ifname patterns on retrieving devices

Only treat the given identifier as Linux netdev name if we can find a
corresponding entry in the device info cache and do not consider strings
starting with "wlan", "ath" or "wl" to be existing devices.

This fixes incorrectly adding wireless sections as ifnames to network
interfaces when the wifi-iface section name begins with one of the
`iface_patterns_wireless` patterns.

Fixes: #5069

(cherry picked from commit d4092b15ce92cda2d7711b57b2251ebebef32b5c)
commit 8bd4e78f (2021-11-10): luci-base: network.js: ignore wireless ifname patterns on retrieving devices

Only treat the given identifier as Linux netdev name if we can find a
corresponding entry in the device info cache and do not consider strings
starting with "wlan", "ath" or "wl" to be existing devices.

This fixes incorrectly adding wireless sections as ifnames to network
interfaces when the wifi-iface section name begins with one of the
`iface_patterns_wireless` patterns.

Fixes: #5069

(cherry picked from commit d4092b15ce92cda2d7711b57b2251ebebef32b5c)