openwrt/luci #5077: System logging does not log to host

kinekt4, 2021-05-28:
In the Luci interface under `System` > `Logging` there is a form field for "External system log server".

`http://[openwrt-ip]/cgi-bin/luci/admin/system/system`

Even though the text hint indicates an IP, this field is free-text and a hostname can be entered.

When the form is saved (successfully) however, logging does not occur.

The reason for this is in: `/etc/config/system` the value is stored under the param `option log_ip` instead of `option log_host`

The form should allow the option for hostnames to be entered such that the hostname is stored in `option log_host` instead of `option log_ip`

jow-, 2021-05-28:
This is actually a bug in OpenWrt. The underlying remote log process accepts hostnames, but the init script validation is wrong.

jow-, 2021-05-28:
Fixed via the commits mentioned above.

closed 2021-05-28 by jow-

kinekt4, 2021-05-28:
Actually. When I manually put hostname in `log_ip` then run `/etc/init.d/log restart` I get `validation failed` error.

> ~# /etc/init.d/log restart
> system.cfg01e48a.log_file is unset and defaults to string (null)
> system.cfg01e48a.log_size=64 validates as uinteger with true
> system.cfg01e48a.log_hostname is unset and defaults to string (null)
> system.cfg01e48a.log_ip=[hostname] validates as ipaddr with false
> system.cfg01e48a.log_remote is unset and defaults to bool 1
> system.cfg01e48a.log_port=[port] validates as port with true
> system.cfg01e48a.log_proto=tcp validates as or("tcp", "udp") with true
> system.cfg01e48a.log_trailer_null is unset and defaults to bool 0
> system.cfg01e48a.log_prefix is unset and defaults to string (null)
> validation failed

The error doesn't show in luci.

kinekt4, 2021-05-28:
OK. Will test when released. Thanks