openwrt/luci #5081: cannot change luci network device macvlan device name

ptpt52, 2021-05-30:
[image]

ptpt52, 2021-05-30:
the device name for macvlan cannot change

jow-, 2021-06-01:
This is intentional. After creating a device configuration, options changing the device identity cannot be changed anymore. If you really want to change the name, you need to reset the configuration and recreate it.

closed 2021-06-01 by jow-

ptpt52, 2021-06-01:
@jow-  not really true
I create a new macvlan and name it as `mvlan01`
after I try to edit, it display `eth0.1mac2`, not the `mvlan01`

ptpt52, 2021-06-01:
it is easy to test
1. create new macvlan and name it to somewhat `xxxx`
2. save and then reopen edit, it display the wrong name.

commit 671c94b0 (2021-06-01): luci-mod-network: don't accidentally reset macvlan device names

Fixes: #5081
commit 5bdb5fd6 (2021-06-12): luci-mod-network: don't accidentally reset macvlan device names

Fixes: #5081

(cherry picked from commit 671c94b034db9d7470d5f1e681b919158de205f6)