openwrt/luci #5084: luci-mod-network: bridge configuration breaks existing config #5068

ezplanet, 2021-05-30:
Issue #5068 is not resolved and it should be re-opened.

ezplanet, 2021-05-30:
With this change a firmware upgrade with consequent acceptance of the luci configuration update in Network Interfaces breaks router functionality. WiFi connected devices no longer communicate with the bridged LAN. As a consequence WiFi connected devices do not receive configuration via DHCP and cannot communicate with the internet.

The only resolution is to restore the previously saved configuration before re-flashing the router with an earlier firmare version. In my case my previous build

Firmware Version | OpenWrt 21.02-SNAPSHOT r16090-bbbc01ede5 / LuCI openwrt-21.02 branch git-21.124.26172-1a59584

Is working fine. The following build r16120 is broken.

ezplanet, 2021-05-31:
Is anyone looking into this?

rmilecki, 2021-05-31:
Really, we don't have crystal balls. Provide `/etc/config/network` and `/etc/config/wireless` from **before** and **after** upgrade + migration.

valdesvj, 2021-05-31:
@ezplanet take a look https://github.com/openwrt/luci/pull/4307

ezplanet, 2021-06-01:
@rmilecki sorry, my routers are all live and I had to revert quickly because of loss of service. 
Should you not run through regression tests before applying such an important change?

rmandrad, 2021-06-03:
ok here's my network; I will not run the upgrade again has it soft bricks the router  

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix ' '

config interface 'lan'
        option type 'bridge'
        option ifname 'lan1 lan2 lan3 lan4'
        option proto 'static'
        option ipaddr '<ipaddr>'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option ifname 'wan'
        option proto 'dhcp'
        option peerdns '0'
        option mtu '9000'

config device 'wan_wan_dev'
        option name 'wan'
        option macaddr ' '

config interface 'wan6'
        option ifname 'wan'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'auto'
        option peerdns '0'
        option mtu '9000'

config interface 'vpnserver'
        option proto 'none'
        option ifname 'ovpns0'

my wireless
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option bursting '1'
        option ff '1'
        option compression '1'
        option country 'FR'
        option cell_density '0'
        option channel '100'
        option htmode 'VHT160'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option key ' '
        option wpa_disable_eapol_key_retries '1'
        option tdls_prohibit '1'
        option encryption 'psk2+ccmp'
        option ssid ' '
 
 

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11g'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option bursting '1'
        option ff '1'
        option compression '1'
        option country 'FR'
        option cell_density '0'
        option htmode 'HT40'
        option channel '9'
        option noscan '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid ''
        option key ''
        option wpa_disable_eapol_key_retries '1'
        option encryption 'psk2+ccmp'
        option max_inactivity '900'
        option macaddr 'mac'

wulfy23, 2021-06-03:
this appears to be before migration config only... and not both as requested...

running non-stable release or master semi implies the ability/responsibility to be able to gather meaningful debugging info and recover from 'soft-bricks'...

while the report is still appreciated... in future it would be even better if you are able to provide the necessary info in order to validate/debug

rmandrad, 2021-06-07:
it appears from my perspective the problem is solved - I applied a number of PRs (mac80211 + Hostapd) that may have fixed the issue of migrating to the new config

rmilecki, 2021-06-09:
@ezplanet: yes. If there is still some issue at all, we need more info to understand it and fix it. OpenWrt is simply an open source project without full time testers or a farm of devices for regression testing.

I've just reviewed https://github.com/openwrt/luci/issues/5068 and explained what has happened there. That ticket is closed for a good reason.

jow-, 2021-08-31:
Seems fixed in recent versions.

closed 2021-08-31 by jow-