openwrt/luci #5085: luci-proto-vpnc --- target network wrong syntax + defaultroute field missing

phqzgunsfjror, 2021-05-30:
**Btw: everything works on the stable release on fritzbox 4040**
Powered by LuCI openwrt-19.07 branch (git-21.044.30835-34e0d65) 
OpenWrt 19.07.7 r11306-c4a6851c72 

**The following issues occur on:**
Powered by LuCI Master (git-21.124.24916-0faf9a4) 
ArgonTheme v2.2.4 
OpenWrt SNAPSHOT r16707-e57e460dc7 
I use it on a raspberry pi 4b.

## How to reproduce:
- install
  - vpnc
  - vpnc-scripts
  - luci-proto-vpnc
- &rarr; network &rarr; interfaces &rarr; add new interface... &rarr;
name: xyz
protocol: VPNC
&rarr; create interface
target network: 192.168.178.0/24

## Problem 1 - target network wrong syntax

When typing `192.168.178.0/24` into field "Target network" there is the error message: 
"Expecting: valid UCI identifier, hostname or IP address"

According to documentation of the plugin the following should be valid:
https://github.com/openwrt/packages/tree/master/net/vpnc
`option target_network '0.0.0.0/0.0.0.0' # network/netmask or CIDR`

So e.g.:
- 192.168.178.0/24
- 192.168.178.0/255.255.255.0

result should be:
```
cat /etc/config/network

config interface 'xyz'
	option proto 'vpnc'
        ...
	option target_network '192.168.178.0/24'
```

## Problem 2 - no field for "option defaultroute"
there is no option to turn off the default route (the default route creates a route for 0.0.0.0 - so everything is routed though the tunnel)

result should be:
```
cat /etc/config/network

config interface 'xyz'
	option proto 'vpnc'
        ...
	option defaultroute '0'
```

closed 2021-05-31 by jow-

jow-, 2021-05-31:
The `target_network` validation issue has been fixed with the commits linked above.

The `defaultroute` option can be found in the advanced tab:
[image]

phqzgunsfjror, 2021-05-31:
Thanks a lot!

commit dece9b06 (2021-05-31): luci-base: fix `network` data type validation

Fixes: #5085
commit 8e14a086 (2021-05-31): luci-base: fix `network` data type validation

Fixes: #5085

(cherry picked from commit dece9b063c16790df731a08f8e73978c51b6c90d)