openwrt/luci #5087: Option to display OVPN log in Luci app

beko136, 2021-05-31:
Hello,
I'd like to suggest adding an option to get OVPN log displayed in Luci app. This could be very helpful in analyzing OVPN issues from a remote location.
Thank you and regards

stangri, 2021-05-31:
@beko136 what's the benefit of having it in the openvpn luci app versus the system log where all the logged/error messages go?

beko136, 2021-05-31:
Hello, justified question, but OVPN on my site doesn't log into syslog. I didn't since I first implemented it, so I was of the opinion, that logging into a separate log file was default behaviour. I'm not aware of any switch in OVPN configuration to activate syslog logging and furthermore separate log files do have their advantages. So I suggested adding the mentioned option into Luci app. Regards