openwrt/luci #5088: nonwildcard (bind dynamic) apply settings broken.

necro-nemesis, 2021-05-31:
-If I apply nonwildcard through uci the entry shows up in both LuCi and /etc/config/dhcp `option nonwildcard '1'` but doesn't have any effect.
-If I then through LuCi uncheck & apply then recheck & apply nonwildcard through LuCi it takes effect.
-If there is no nonwildcard set through uci and I check it in LuCi and apply no entry is added to /etc/config/dhcp.

Something seems borked between LuCi and uci handling nonwildcard settings and actually applying them correctly.

FWIW I'm using DNSMasq full package although I believe that to be unrelated.

jow-, 2021-05-31:
[According to `/etc/init.d/dnsmasq`](https://github.com/openwrt/openwrt/blob/master/package/network/services/dnsmasq/files/dnsmasq.init#L887), the `nonwildcard` option defaults to `1` if unset, LuCI follows this logic. If you tick the checkbox, the option is removed from `/etc/config/dhcp`. If you untick the checkbox, `option nonwildcard 0` is written to `/etc/config/dhcp`.

If you manually add `option nonwildcard 1` and save with LuCI, the option will likely get removed (as it defaults to `1`) but it should have no effect.

### Testing

#### Without any `option nonwildcard`:

```
root@OpenWrt:~# uci show dhcp.@dnsmasq[0]
dhcp.cfg01411c=dnsmasq
dhcp.cfg01411c.domainneeded='1'
dhcp.cfg01411c.localise_queries='1'
dhcp.cfg01411c.rebind_protection='1'
dhcp.cfg01411c.rebind_localhost='1'
dhcp.cfg01411c.local='/lan/'
dhcp.cfg01411c.domain='lan'
dhcp.cfg01411c.expandhosts='1'
dhcp.cfg01411c.authoritative='1'
dhcp.cfg01411c.readethers='1'
dhcp.cfg01411c.leasefile='/tmp/dhcp.leases'
dhcp.cfg01411c.resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.cfg01411c.localservice='1'
dhcp.cfg01411c.ednspacket_max='1232'
root@OpenWrt:~# 
```

`bind-dynamic` is enabled:

```
root@OpenWrt:~# grep bind-dynamic /var/etc/dnsmasq.conf.*
bind-dynamic
```

#### With `nonwildcard` disabled:

```
root@OpenWrt:~# uci set dhcp.@dnsmasq[0].nonwildcard=0
root@OpenWrt:~# uci show dhcp.@dnsmasq[0]
dhcp.cfg01411c=dnsmasq
dhcp.cfg01411c.domainneeded='1'
dhcp.cfg01411c.localise_queries='1'
dhcp.cfg01411c.rebind_protection='1'
dhcp.cfg01411c.rebind_localhost='1'
dhcp.cfg01411c.local='/lan/'
dhcp.cfg01411c.domain='lan'
dhcp.cfg01411c.expandhosts='1'
dhcp.cfg01411c.authoritative='1'
dhcp.cfg01411c.readethers='1'
dhcp.cfg01411c.leasefile='/tmp/dhcp.leases'
dhcp.cfg01411c.resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.cfg01411c.localservice='1'
dhcp.cfg01411c.ednspacket_max='1232'
dhcp.cfg01411c.nonwildcard='0'
root@OpenWrt:~# 
```

`bind-dynamic` is disabled:

```
root@OpenWrt:~# /etc/init.d/dnsmasq restart
udhcpc: started, v1.33.1
udhcpc: sending discover
udhcpc: no lease, failing
root@OpenWrt:~# grep bind-dynamic /var/etc/dnsmasq.conf.*
root@OpenWrt:~# 
```

#### With `nonwildcard` enabled:

```
root@OpenWrt:~# uci set dhcp.@dnsmasq[0].nonwildcard=1
root@OpenWrt:~# uci show dhcp.@dnsmasq[0]
dhcp.cfg01411c=dnsmasq
dhcp.cfg01411c.domainneeded='1'
dhcp.cfg01411c.localise_queries='1'
dhcp.cfg01411c.rebind_protection='1'
dhcp.cfg01411c.rebind_localhost='1'
dhcp.cfg01411c.local='/lan/'
dhcp.cfg01411c.domain='lan'
dhcp.cfg01411c.expandhosts='1'
dhcp.cfg01411c.authoritative='1'
dhcp.cfg01411c.readethers='1'
dhcp.cfg01411c.leasefile='/tmp/dhcp.leases'
dhcp.cfg01411c.resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.cfg01411c.localservice='1'
dhcp.cfg01411c.ednspacket_max='1232'
dhcp.cfg01411c.nonwildcard='1'
root@OpenWrt:~# 
```

`bind-dynamic` is enabled:

```
root@OpenWrt:~# /etc/init.d/dnsmasq restart
udhcpc: started, v1.33.1
udhcpc: sending discover
udhcpc: no lease, failing
root@OpenWrt:~# grep bind-dynamic /var/etc/dnsmasq.conf.*
bind-dynamic
root@OpenWrt:~# 
```

jow-, 2021-05-31:
Works for me.

closed 2021-05-31 by jow-

necro-nemesis, 2021-05-31:
Is it "working"?  I appreciate the default condition is on if there's no entry but with `dnsmasq full` the only time I have found it has taken effect ( the most critical part ) is cycling the gui from off apply to on apply.

jow-, 2021-05-31:
It is working insofar as that the option gets propagated to uci and from there to the generated native dnsmasq configuration. If the dnsmasq restart or config reload is somehow flaky then this is outside of the scope of the LuCI ui.

necro-nemesis, 2021-05-31:
100% agree. LuCi is doing what it's supposed to and I looked over /etc/init.d/dnsmasq in terms of what it's intent is with the service but there appears to be flakiness at the dnsmasq full package area. It's getting fed the correct settings, it just appears to be temperamental in initially applying them. Thanks for your insight.