openwrt/luci #5089: luci-proto-qmi fail to setup via luci web

ptpt52, 2021-06-01:
since upstream https://github.com/openwrt/openwrt/commit/4b9a67362d70c544b85078b8d5c661f43f7472d9
it use `device` to replace `ifname`, this break the qmi

qmi need to set option
```
option device /dev/cdc-wmm
```
for works

rmilecki, 2021-06-01:
Show you `/etc/board.json` and `/etc/config/network` right after clean flash / firstboot.

ptpt52, 2021-06-01:
> Show you `/etc/board.json` and `/etc/config/network` right after clean flash / firstboot.

it is fresh new. I do not use old config

rmilecki, 2021-06-01:
Show you `/etc/board.json` and `/etc/config/network` right after fresh new install.

ptpt52, 2021-06-01:
[image]

I post an img for this new version of now, where setup fails

rmilecki, 2021-06-01:
Please execute two following commands:
* `cat /etc/board.json`
* `cat /etc/config/network`

and copy & paste here the output

ptpt52, 2021-06-01:
as we now use `device` to replace  `ifname`, this is the conflict.

the working config for qmi setup in old version:
```
config interface 'ltewan'
	option proto 'qmi'
	option device '/dev/cdc-wdm0'
	option auth 'none'
	option pdptype 'ipv4v6'
```

and the new config if I do the same setup via luci web

```
config interface 'ltewan'
	option proto 'qmi'
	option auth 'none'
	option pdptype 'ipv4v6'
```

the /etc/board.json:
```
{
	"model": {
		"id": "xwrt,mac500f",
		"name": "MAC500F"
	},
	"network": {
		"lan": {
			"ports": [
				"lan1",
				"lan2",
				"lan3",
				"lan4"
			],
			"protocol": "static"
		},
		"wan": {
			"device": "wan",
			"protocol": "dhcp"
		}
	},
	"system": {
		"compat_version": "1.1"
	}
}
```

the /etc/config/network
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'
	option ula_prefix 'fda2:e9fa:f6fa::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.15.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'

config interface 'usbwan'
	option proto 'dhcp'
	option device 'usb0'
	option metric '800'

config interface 'ltewan'
	option proto 'qmi'
	option auth 'none'
	option pdptype 'ipv4v6'

```

ptpt52, 2021-06-01:
I am sure same issue with `luci-proto-ncm`, `luci-proto-3g`, `luci-proto-mbim`

ptpt52, 2021-06-01:
The root cause is the modification of netifd, which uses device instead of ifname, we may update and modify the module of luci-proto-xxx simultaneously

rmilecki, 2021-06-01:
OK, a little step back. There isn't anything `qmi` related in your `/etc/board.json`, so https://github.com/openwrt/openwrt/commit/4b9a67362d70c544b85078b8d5c661f43f7472d9 has nothing to do with this issue. `config_generate` takes care of "translating" `/etc/board.json` into `/etc/config/network`.

There is a problem with QMI protocol implementation in `netifd` and/or LuCI. I'll take a look at it.

ptpt52, 2021-06-01:
@rmilecki  thanks, just as what you tell

jow-, 2021-06-01:
Fixed by https://github.com/openwrt/luci/commit/f3f74bd0fe66b94a99a8d944b63dcd6bdd1b93c6, https://github.com/openwrt/luci/commit/10f02472c5fdab72615a7d3695e8f354811cd661 and https://github.com/openwrt/luci/commit/96ee6dc8d6c9796ab67de6f313a068a4bd3bb20f

closed 2021-06-01 by jow-

sebastianertz, 2021-06-02:
@jow- The problem also exists in the branch openwrt-21.02

martin8801, 2021-06-14:
Will this make it in to 21.02?

rmilecki, 2021-06-18:
It should be fixed in the 21.02-rc3