openwrt/luci #5090: luci-mod-network: device page, reset buttong, which deletes, is misleading

pmelange, 2021-06-01:
While testing rc2, I tried setting up some more complicated bridges and veth devices.  While experimenting I noticed the "reset" button, and thought I would give it a try.  I didn't notice until too late (the mouseover text was visible just as I was clicking) that the device I had set up would be deleted.

I recommend that the name "reset" be changed to "remove" (like in wireless) or "delete" (like in switch and firewall).

jow-, 2021-06-01:
Neither remove nor delete are correct labels since that button just clears the `config device` configuration. The related netdev is managed by netifd and might or might not disappear. Fixed netdevs like `eth0` or `eth1` cannot be "removed" or "deleted" at all. You can just reset their configuration to default.

closed 2021-06-01 by jow-

pmelange, 2021-06-01:
This doesn't change the fact that the mouse-over show "delete".

Just to verify, I created a new device of type bridge with two ports.  Then I saved and applied the changes.  The device section was added to /etc/config/network.  Then when clicking "reset" and followed by save and apply, the device section was removed.  

Clicking on "reset" does actually remove the device section from /etc/config/network.  Please consider reopening this issue so that a better button label can be used.

jow-, 2021-06-01:
> This doesn't change the fact that the mouse-over show "delete".
Your ticket title stated "reset button label"

> Clicking on "reset" does actually remove the device section from /etc/config/network.
No, it stages a deletion of the device section.

I'll change the tooltip.

SvenRoederer, 2021-07-15:
Based on my PR in #5092 and the discussion here, the proper term fir the label might be "unconfigure".
AS mentioned by @jow- the physical device will stay in the system, but also "reset" implies that the "non-default changes" of the device will be reverted / defaulted. Essentially that's what the changed tooltip in 26e21d4 also tells.

SvenRoederer, 2021-07-15:
> the proper term for the label might be "unconfigure".

PR5094 is made.