openwrt/luci #5092: Access Multiple Luci behind nginx reverse proxy

valm13, 2021-06-01:
Hi,

I'm trying to create access to multiple OpenWrt Luci through Nginx reverse proxy.

I want to access them with a specif URL, providing the IP of the remote machine: /target-ip/[IP].

Here is the related question I've asked on StackOverflow: https://stackoverflow.com/questions/67787122/multiple-openwrt-luci-behind-nginx-reverse-proxy 

PS: I'm gonna recreate the question on SuperUser

jow-, 2021-06-01:
You could try modifying the `lua_prefix` and `ubus_prefix` options of `/etc/config/uhttpd` but I suppose it is easier to do use nginx' `sub_filter` facility to rewrite embedded resource links as needed. See https://serverfault.com/questions/713148/modify-html-pages-returned-by-nginx-reverse-proxy for examples.

closed 2021-06-01 by jow-

valm13, 2021-06-24:
Thanks for the tip ! Unfortunately, I've tried it, and it didn't work : I was only able to reach the login page
I've found another way to bypass this problem, simply by creating a subdomain for each Luci web app.