openwrt/luci #5104: Stops at login page [WRT3200ACM]

wrtechnopath, 2021-06-06:
I started with the stock firmware completely cleared and under the connectivity page I upgraded the firmware using openwrt-21.02.0-rc2-wrt3200acm-squashfs-factory.img

Everything looked like it took fine.

However, when I try using http://192.168.1.1 I'm redirected to http://192.168.1.1/cgi-bin/luci/. The warning says that no password is installed, so I press the login button and nothing happens.

I'm able to ```ssh / scp``` to the router

jow-, 2021-06-07:
This is usually caused by browser heuristics. If LuCI's "sysauth" cookie was once set via HTTPS in the past, browsers refuse to accept a new "sysauth" cookie via plain HTTP. Due to that you're thrown back to the login dialog after pressing login.

For Firefox:

- press F12 to open the inspector
- select the "Storage" tab, open the cookies list on the left side
- select the "sysauth" cookie row on the right and press delete to remove it
- close the inspector and retry

For Chrome:

- press F12 to open the inspector
- select the "Application" tab
- on the left side, open the "Cookies" list and select the "http://192.168.1.1" origin
- select the "sysauth" cookie row on the right and press delete to remove it
- close the inspector and retry

closed 2021-06-08 by wrtechnopath

richb-hanover, 2021-07-31:
Thanks @jow for digging out this diagnostic info. Let me see if I understand the problem:

- I flash over factory firmware of Router A. This sets a sysauth cookie.
- Maybe I enable https on Router A, so the sysauth cookie would be set with https.
- Five years later, I get a new Router B, flash over factory *its* factory firmware. 
- Now I'm in the bad state. My browser remembers the https sysauth cookie from Router A (on the default 192.168.1.1 address), but my connection to Router B (http only, same address) ignores the new sysauth cookie

Do I have that right? If so, we really need to find a solution because the error message presented is simply wrong. It essentially says, "You have the wrong name/password" when in fact no password can ever be used to log in.

And for some possible solutions:

- Does the sysauth cookie have an expiration date?
- What might it mean to have explicit sysauth-http and sysauth-https cookies? Perhaps they could help identify this state...

Thanks for listening.

shodanx2, 2021-09-14:
Hello, 

I just upgraded from version 18, clean wipe.
Has the same issue as OP. 
Try to login with good password, there was no error it just didn't log in, same as pressing F5.
I could login with a firefox private window
Deleting sysauth cooking as outlined above as fixed the issue

jow-, 2022-02-16:
Then your browser is not accepting the `sysauth` cookie for whatever reason. Maybe security heuristics.

richb-hanover, 2022-02-16:
When I get into this state, I have resigned myself to erasing cookies for 192.168.1.1 (or whatever router address). That immediately clears the problem.

neoOpus, 2022-03-01:
I confirm, clearing the cookie, reloading resolves the issue