openwrt/luci #5106: Channel option not appearing in 5ghz radio

CaboBeach, 2021-06-08:
There is no channel option in Luci in the 5ghz radio when selecting AC mode, although it works fine when setting in cli, channel option is still there in N and Legacy modes. Netgear R7800/OpenWrt SNAPSHOT r16906-f36990eae7 / LuCI Master git-21.158.53317-da97288

[image]

CaboBeach, 2021-06-08:
Thanks @zxlhhyccc, didn't realize this was already closed, i'll go ahead and close this as well.

zxlhhyccc, 2021-06-08:
> Thanks @zxlhhyccc, didn't realize this was already closed, i'll go ahead and close this as well.

Hello, problem not solved why close it?

CaboBeach, 2021-06-08:
Just assuming the devs were aware and working on it if it had been closed, I’ll go ahead and reopen.

hnyman, 2021-06-08:
Works for me.
R7800 with master r16914-1aa80ce393  with LuCI  git-21.156.61451-516c58e

[image]

zxlhhyccc, 2021-06-08:
> Works for me.
> R7800 with master r16914-1aa80ce393 with LuCI git-21.156.61451-516c58e
> 
> [image]

You are not compiling firmware using the master branch

zxlhhyccc, 2021-06-08:
Can be flexible solution, namely: first set N mode to save and apply and then select AC mode can set the channel

> There is no channel option in Luci in the 5ghz radio when selecting AC mode, although it works fine when setting in cli, channel option is still there in N and Legacy modes. Netgear R7800/OpenWrt SNAPSHOT r16906-f36990eae7 / LuCI Master git-21.158.53317-da97288
> 
> [image]

https://github.com/openwrt/luci/issues/5106#issue-914125730

hnyman, 2021-06-08:
> > Works for me.
> > R7800 with master r16914-1aa80ce393 with LuCI git-21.156.61451-516c58e

> You are not compiling firmware using the master branch

No idea what you mean.
This is a normal clean master build. Commits are right off the master branch, both are the current HEADs:
* main OpenwWrt: https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=1aa80ce3931736f1020f69f486a815c2b2575d19
* LuCI: https://git.openwrt.org/?p=project/luci.git;a=shortlog;h=da97288015e0a8919c55075d71d88890e2f339f3

zxlhhyccc, 2021-06-08:
> > You are not compiling firmware using the master branch
> 
> No idea what you mean.
> This is a normal clean master build. Commits are right off the master branch, both are the current HEADs:
> EDIT:
> 
> ```
> main      2021-06-07 1aa80ce scripts/feeds: generate index after all f
> luci      2021-06-07 da97288 Translated using Weblate (Chinese (Simpli
> ```
> 
> (the version shown in my previous message was footer, from luci-base, but the build is up-to-date)

I also have clean firmware, but the same problem as the first floor

jow-, 2021-06-08:
Please provide the output of the following cli commands:
```
ubus call iwinfo info '{ "device": "radio0" }'
ubus call iwinfo info '{ "device": "radio1" }'
ubus call iwinfo freqlist '{ "device": "radio0" }'
ubus call iwinfo freqlist '{ "device": "radio1" }'
```

hnyman, 2021-06-08:
@jow- 
Here is reference data from my own R7800.
I use the default -ct ath10k driver.

```
 OpenWrt SNAPSHOT, r16914-1aa80ce393
 -----------------------------------------------------
root@router1:~# opkg list-installed | grep ath10
ath10k-board-qca9984 - 20210511-1
ath10k-firmware-qca9984-ct - 2020-11-08-1
kmod-ath10k-ct - 5.4.124+2021-06-03-b44cd7b2-1

root@router1:~# ubus call iwinfo info '{ "device": "radio0" }'
{
        "phy": "phy0",
        "bssid": "DC:EF:...",
        "country": "FI",
        "mode": "Master",
        "channel": 36,
        "center_chan1": 38,
        "frequency": 5180,
        "frequency_offset": 0,
        "txpower": 23,
        "txpower_offset": 0,
        "quality_max": 70,
        "noise": 0,
        "htmodes": [
                "HT20",
                "HT40",
                "VHT20",
                "VHT40",
                "VHT80",
                "VHT80+80",
                "VHT160"
        ],
        "hwmodes": [
                "ac",
                "n"
        ],
        "hwmode": "n",
        "htmode": "HT40",
        "hardware": {
                "id": [
                        5772,
                        70,
                        5772,
                        51966
                ],
                "name": "Qualcomm Atheros QCA9984"
        }
}
root@router1:~# ubus call iwinfo info '{ "device": "radio1" }'
{
        "phy": "phy1",
        "bssid": "DC:EF:...",
        "country": "FI",
        "mode": "Master",
        "channel": 3,
        "center_chan1": 3,
        "frequency": 2422,
        "frequency_offset": 0,
        "txpower": 20,
        "txpower_offset": 0,
        "quality_max": 70,
        "noise": 0,
        "htmodes": [
                "HT20",
                "HT40"
        ],
        "hwmodes": [
[... 242 lines cut ...]
                        "mhz": 2457,
                        "restricted": false,
                        "active": false
                },
                {
                        "channel": 11,
                        "mhz": 2462,
                        "restricted": false,
                        "active": false
                },
                {
                        "channel": 12,
                        "mhz": 2467,
                        "restricted": false,
                        "active": false
                },
                {
                        "channel": 13,
                        "mhz": 2472,
                        "restricted": false,
                        "active": false
                }
        ]
}
```

jow-, 2021-06-08:
I don't fully recall the logic but the channel selector should only be hidden if there's less than 2 channels available. Since AC mode implies 5GHz I suspect that people observing this issue have no 5GHz channels available [according to iwinfo data]. It might not actually be a LuCI issue but could be related to https://bugs.openwrt.org/index.php?do=details&task_id=3856

CaboBeach, 2021-06-08:
I'm currently using the non-ct driver (can install snapshot with -ct driver in a few hours to test if necessary)

```
:~# ubus call iwinfo info '{ "device": "radio0" }'
{
	"phy": "phy0",
	"bssid": "DC:EF:09:E3:B8:68",
	"country": "US",
	"mode": "Master",
	"channel": 153,
	"center_chan1": 155,
	"frequency": 5765,
	"frequency_offset": 0,
	"txpower": 30,
	"txpower_offset": 0,
	"quality_max": 70,
	"noise": 0,
	"htmodes": [
		"HT20",
		"HT40",
		"VHT20",
		"VHT40",
		"VHT80",
		"VHT80+80",
		"VHT160"
	],
	"hwmodes": [
		"ac",
		"n"
	],
	"hwmode": "ac",
	"htmode": "VHT80",
	"hardware": {
		"id": [
			5772,
			70,
			5772,
			51966
		],
		"name": "Qualcomm Atheros QCA9984"
	}
}

```

```
~# ubus call iwinfo info '{ "device": "radio1" }'
{
	"phy": "phy1",
	"bssid": "DC:EF:09:E3:B8:69",
	"country": "US",
	"mode": "Master",
	"channel": 6,
	"center_chan1": 6,
	"frequency": 2437,
	"frequency_offset": 0,
	"txpower": 30,
	"txpower_offset": 0,
	"quality_max": 70,
	"noise": 0,
	"htmodes": [
		"HT20",
		"HT40"
	],
	"hwmodes": [
		"b",
		"g",
		"n"
	],
	"hwmode": "n",
	"htmode": "HT20",
	"hardware": {
		"id": [
			5772,
			70,
			5772,
			51966
		],
		"name": "Qualcomm Atheros QCA9984"
	}
}
```

```
~# ubus call iwinfo freqlist '{ "device": "radio0" }'
{
	"results": [
		{
			"channel": 36,
			"mhz": 5180,
			"restricted": false,
			"active": false
		},
		{
			"channel": 40,
			"mhz": 5200,
			"restricted": false,
			"active": false
		},
		{
			"channel": 44,
			"mhz": 5220,
			"restricted": false,
			"active": false
		},
		{
			"channel": 48,
			"mhz": 5240,
			"restricted": false,
			"active": false
		},
		{
			"channel": 52,
			"mhz": 5260,
			"restricted": false,
			"active": false
		},
		{
			"channel": 56,
			"mhz": 5280,
			"restricted": false,
			"active": false
		},
		{
			"channel": 60,
			"mhz": 5300,
			"restricted": false,
			"active": false
		},
		{
			"channel": 64,
			"mhz": 5320,
			"restricted": false,
			"active": false
		},
		{
			"channel": 100,
			"mhz": 5500,
			"restricted": false,
			"active": false
		},
		{
			"channel": 104,
			"mhz": 5520,
			"restricted": false,
			"active": false
		},
		{
			"channel": 108,
			"mhz": 5540,
			"restricted": false,
			"active": false
		},
		{
			"channel": 112,
			"mhz": 5560,
			"restricted": false,
			"active": false
		},
		{
			"channel": 116,
			"mhz": 5580,
			"restricted": false,
			"active": false
		},
		{
			"channel": 120,
			"mhz": 5600,
			"restricted": false,
			"active": false
		},
		{
			"channel": 124,
			"mhz": 5620,
			"restricted": false,
			"active": false
		},
		{
			"channel": 128,
			"mhz": 5640,
			"restricted": false,
			"active": false
		},
		{
			"channel": 132,
			"mhz": 5660,
			"restricted": false,
			"active": false
		},
		{
			"channel": 136,
			"mhz": 5680,
			"restricted": false,
			"active": false
		},
		{
			"channel": 140,
			"mhz": 5700,
			"restricted": false,
			"active": false
		},
		{
			"channel": 144,
			"mhz": 5720,
			"restricted": false,
			"active": false
		},
		{
			"channel": 149,
			"mhz": 5745,
			"restricted": false,
			"active": false
		},
		{
			"channel": 153,
			"mhz": 5765,
			"restricted": false,
			"active": true
		},
		{
			"channel": 157,
			"mhz": 5785,
			"restricted": false,
			"active": false
		},
		{
			"channel": 161,
			"mhz": 5805,
			"restricted": false,
			"active": false
		},
		{
			"channel": 165,
			"mhz": 5825,
			"restricted": false,
			"active": false
		}
	]
}
```

```
~# ubus call iwinfo freqlist '{ "device": "radio1" }'
{
	"results": [
		{
			"channel": 1,
			"mhz": 2412,
			"restricted": false,
			"active": false
		},
		{
			"channel": 2,
			"mhz": 2417,
			"restricted": false,
			"active": false
		},
		{
			"channel": 3,
			"mhz": 2422,
			"restricted": false,
			"active": false
		},
		{
			"channel": 4,
			"mhz": 2427,
			"restricted": false,
			"active": false
		},
		{
			"channel": 5,
			"mhz": 2432,
			"restricted": false,
			"active": false
		},
		{
			"channel": 6,
			"mhz": 2437,
			"restricted": false,
			"active": true
		},
		{
			"channel": 7,
			"mhz": 2442,
			"restricted": false,
			"active": false
		},
		{
			"channel": 8,
			"mhz": 2447,
			"restricted": false,
			"active": false
		},
		{
			"channel": 9,
			"mhz": 2452,
			"restricted": false,
			"active": false
		},
		{
			"channel": 10,
			"mhz": 2457,
			"restricted": false,
			"active": false
		},
		{
			"channel": 11,
			"mhz": 2462,
			"restricted": false,
			"active": false
		}
	]
}
```

castillofrancodamian, 2021-06-08:
> I don't fully recall the logic but the channel selector should only be hidden if there's less than 2 channels available. Since AC mode implies 5GHz I suspect that people observing this issue have no 5GHz channels available [according to iwinfo data]. It might not actually be a LuCI issue but could be related to https://bugs.openwrt.org/index.php?do=details&task_id=3856
> 
> The linked issue is caused by mac80211.sh getting new logic for handling the `option path` selector while iwinfo was not updated accordingly to use the new logic. This currently affects all multi phy radios. No idea if Netgear R7800 uses one of these.

It is not true. I have an Archer C60 that has AC Wi-Fi.

CaboBeach, 2021-06-08:
Made a fresh build with ct drivers and the channel option for AC is still missing on OpenWrt SNAPSHOT r16915-bb749e49a6 / LuCI Master git-21.158.53317-da97288

[image]

[image]

hnyman, 2021-06-09:
You are right, the bug is there. 
But looks like it is not fully a LuCI bug.

@jow- 
If I remove the current wifi config, do "wifi detect" to generate the default config, and then go to LuCI, the field is empty.
The bug is likely actually outside LuCI: the generated default wifi config files does not set any "channel" parameter for the 5GHz radio in R7800.

Failing default config below. Note that the 2 GHz radio has "channel 1", but the 5 GHz radio is totally missing the option:

```
root@router1:/etc/config# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
```

LuCI GUI just reacts to the missing option and does not allow to set one in AC mode.

hnyman, 2021-06-09:
Actually, even more funny. Tried again. This time there is first the "channel 36" first, and after enabling the radio in LuCI it has been replaced by "channel auto" option and the initial default VHT80 changed to VHT20.

Looks like the file gets modified by LuCI a bit differently on depending if the radio is on or off, and the options are also not fully only dependent on the /etc/config/wireless, but are also affected by the current live radio settings.  (I got first the above wrong analysis, as I had left previous wifi on, but had generated a new default config file.)

hnyman, 2021-06-09:
I analysed again.
Starting from default config, and toggling radio0 to "Enabled" in LuCI removes the AC channel option:
```
root@router1:/etc/config# diff -u wireless.newdefault  wireless.toggled
--- wireless.newdefault 2021-06-09 08:50:21.000000000 +0300
+++ wireless.toggled    2021-06-09 08:52:33.000000000 +0300
@@ -2,10 +2,11 @@
 config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
-       option channel '36'
        option band '5g'
-       option htmode 'VHT80'
-       option disabled '1'
+       option hwmode '11a'
+       option channel 'auto'
+       option htmode 'VHT20'
+       option cell_density '0'

 config wifi-iface 'default_radio0'
        option device 'radio0'
```

[image]

Curiously, even if I remove the "disabled" line with text editor (and leave the channel), the AC channel selection is still missing.

Default config (regarding radio0):
```
root@router1:/etc/config# cat wireless.newdefault

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

```

Toggled to enabled:
``` 
root@router1:/etc/config# cat wireless.toggled2

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

```

debug output (after toggling to Enabled):

[... 199 lines cut ...]
                        "mhz": 5805,
                        "restricted": false,
                        "active": false
                },
                {
                        "channel": 165,
                        "mhz": 5825,
                        "restricted": false,
                        "active": false
                },
                {
                        "channel": 169,
                        "mhz": 5845,
                        "restricted": false,
                        "active": false
                },
                {
                        "channel": 173,
                        "mhz": 5865,
                        "restricted": false,
                        "active": false
                }
        ]
}
```

jow-, 2021-06-09:
>        option band '5g'

That seems to be a new wireless option, never seen that one so far. LuCI expects `option hwmode` to be set to `11a`.

jow-, 2021-06-09:
Please see if PR https://github.com/openwrt/luci/pull/5110 resolves your issue.

hnyman, 2021-06-09:
> OpenWrt master appears to use option band while openwrt-21.02 uses option hwmode yet, but can't rule out that this silent change will sneak in there too through a backport or similar.

Hopefully at least compatibility, or wre will get swamped in support issues. (many people jump between master snapshots and the recent releases)

zxlhhyccc, 2021-06-09:
> Please see if PR #5110 resolves your issue.

Problem solved!

closed 2021-06-09 by jow-

CaboBeach, 2021-06-09:
Problem solved here too, thanks for the help @hnyman and @jow- 

[image]

PR #5110 (2021-06-09): luci-mod-network: support newer wireless.wifi-device.band option
OpenWrt switched from "option hwmode" to "option band" in order to select
the frequency band to use for the radio phy.

Extend the channel selector to recognize and use an existing "option band"
to select the appropriate channel list. When operating upon a wireless
configuration still using "option hwmode", then translate it to a band
value internally and translate it back to "option hwmode" on save.

This should provide forward- and backwards compatibility with both current
OpenWrt master and older versions still using hwmode.

 [... cut]
commit 0829d312 (2021-06-09): luci-mod-network: support newer wireless.wifi-device.band option

OpenWrt switched from "option hwmode" to "option band" in order to select
the frequency band to use for the radio phy.

Extend the channel selector to recognize and use an existing "option band"
to select the appropriate channel list. When operating upon a wireless
configuration still using "option hwmode", then translate it to a band
value internally and translate it back to "option hwmode" on save.

This should provide forward- and backwards compatibility with both current
OpenWrt master and older versions still using hwmode.
 [... cut]