openwrt/luci #5113: Permanent redirect to /cgi-bin/luci is unhelpful

proski, 2021-06-09:
I've been using a router with OpenWrt on 192.168.1.1 until recently. Today I had to configure another router with the same IP address. Whenever I would enter http://192.168.1.1/ Firefox would get me to http://192.168.1.1/cgi-bin/luci that would obviously fail. It took me a white to work around. I had to check on Stackoverflow how to clear the Firefox redirect cache. I believe the redirect was marked as "permanent". Would it be possible to use a temporary redirect instead?

jow-, 2021-06-10:
It is a meta refresh tag, no permanent redirect. The page is already marked as no-cache through various meta tags but browsers cache it anyway. No way around arm.

closed 2021-06-10 by jow-

logan2611, 2021-07-24:
Does https://github.com/openwrt/luci/issues/731#issuecomment-333411707 not work?

golf247, 2023-07-29:
Closed or not closed, this redirect is a major pain that is well beyond what a home user should have to deal with.   I'm putting this here in case helpful to others.   I'm not claiming to have figured it out, as I found on another site.   I was bouncing between freshtomato and openwrt router.  Chrome wanted to redirect back to luci when I tried to access tomato.  

Only solution I found was to go into Chrome developer tools (F12); network tab; disable cache.  Then go to freshtomato routerIP.   After that, unchecked disable cache.   That's the easiest method I've seen for stopping the luci redirect.

Miigon, 2023-09-12:
Here to say this still bugs me to this day

dhewg, 2023-09-12:
To state the obvious: complaining about it for 7 years (!) apparently doesn't help!

Assuming the linked way 3 comments above works it'll take just a few minutes to paste that into modules/luci-base/root/www/index.html, confirming it works and then opening a PR about it.

Miigon, 2023-09-29:
Ok, someone (much less ignorant than me) actually [have done it](https://github.com/openwrt/luci/commit/34dd31ae9a63c573646cce9f75ca5edf4bc4e82e) a year ago. It's mainly a firefox issue now and has been fixed in that commit. (However the current stable OpenWrt release 22.03 ship with a version of luci that does not contain that fix)

For any firefox user that happened to stumble upon this problem, you can either wait for the next stable openwrt release, or just add these two lines into your `/www/index.html`:

```html
<html xmlns="http://www.w3.org/1999/xhtml">
	<head>
		<meta http-equiv="Cache-Control" content="no-cache, no-store, must-revalidate" />
+		<meta http-equiv="Pragma" content="no-cache" />
+		<meta http-equiv="Expires" content="0" />
		<meta http-equiv="refresh" content="0; URL=cgi-bin/luci/" />
		<style type="text/css">
```