openwrt/luci #5125: cbi-checkbox and labels do not wrap cleanly

nickberry17, 2021-06-14:
It looks like these would benefit from being wrapped together so that the text and checkboxes are not separated when it overflows.
[image]

nickberry17, 2021-07-20:
I fixed this by setting the display css property to flex with justify: flex-start. Fixes the problem and does not mess with other checkboxes.  Will send a PR.

adelton, 2025-01-11:
Running `git grep -B 6 'justify.*flex-start'` in the repo checkout does not suggest that the PR was filed, so likely not a high priority for the OP.

And `git grep '4G LTE Bands'` does not find anything so not sure on what page this could be demonstrated, so that I could possibly test and file that PR myself.

I suggest to close unless someone knows about a nice reproducer in the latest OpenWrt versions.

closed 2025-01-11 by systemcrash