openwrt/luci #5126: Wrong hostname/IP in "Associated Stations" list

anyuta1166, 2021-06-14:
Firmware Version: OpenWrt SNAPSHOT r16952-677813c776 / LuCI Master git-21.163.60157-1bcb125

I've found that one Wi-Fi client in "Associated Stations" list is shown with the wrong hostname and IP:

[image]

margarita.lan and 192.168.20.1 is the hostname and the IP address of the router itself.

In "Active DHCP Leases" this client is displayed correctly with IP 192.168.20.106 and no hostname:

[image]

After some investigation I've found that getHostHints luci rcp call returns incorrect info.

The first element in the array returned by this call is
`"12:37:7F:06:96:90":{"ipaddrs":["192.168.20.1","192.168.20.106","192.168.20.113","192.168.20.117","192.168.20.120","192.168.20.122","192.168.20.131","192.168.20.185","192.168.20.189"],"ip6addrs":["fd87:90d1:5d0:0:7c2c:e990:5f4b:b73c","fe80::1037:7fff:fe06:9690"],"name":"margarita.lan"}`
Here we have the MAC address of the Wi-Fi client, list of all (!) IPv4 addresses in local network and router hostname.
All other entries looks good, like this: `"E4:AA:EA:5A:98:5F":{"ipaddrs":["192.168.20.185"],"ip6addrs":["fd87:90d1:5d0:0:90a4:41bb:4251:f8a7","fe80::b53c:80fa:82f6:ab8c"],"name":"Tvillanote"}`

anyuta1166, 2021-06-14:
I've also checked another router for this issue. It has slightly older Firmware Version: OpenWrt SNAPSHOT r16915-bb749e49a6 / LuCI Master git-21.156.61451-516c58e, however as far as I see there were no changes that might affect this issue in the last 5 days.
"Associated Stations" list is totally fine.
The first element in the array returned by getHostHints call is (external IPs redacted for privacy):
`{"00:00:5E:00:01:D5":{"ipaddrs":["95.31.xx.xx","95.31.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","178.140.xx.xx","192.168.0.2","192.168.0.3","192.168.0.107","192.168.0.108","192.168.0.110","192.168.0.112","192.168.0.118","192.168.0.119","192.168.0.120","192.168.0.122","192.168.0.124","192.168.0.126","192.168.0.130","192.168.0.134","192.168.0.142","192.168.0.143","192.168.0.146","192.168.0.159","192.168.0.160","192.168.0.161","192.168.0.164","192.168.0.167","192.168.0.168","192.168.0.169","192.168.0.173","192.168.0.185","192.168.0.186","192.168.0.187","192.168.0.192","192.168.0.196","192.168.0.240","192.168.0.254"],"ip6addrs":[],"name":"tasmota-AB111D.lan"}`
Here we have a MAC address (I don't know whose this MAC address is), a list of some external and internal IPs (probably from ARP table? because I see neighbour external addresses), and a hostname of one of Wi-Fi clients. This Wi-Fi client is shown correctly in the list because there is also an entry with its MAC address:
`"DC:4F:22:AB:11:1D":{"ipaddrs":["192.168.0.168"],"ip6addrs":[],"name":"tasmota-AB111D"}`

So to summarize: the first element returned by getHostHints call has a random (?) MAC address, list of IP addresses (from ARP table?) and a random (?) hostname.
If this MAC address is equal to the MAC address of one of the Wi-Fi clients, this particular Wi-Fi client is shown with the wrong hostname and IP in "Associated Stations" list.

jow-, 2021-06-14:
Does the behavior persist if you move any of the following files away?

- /etc/config/dhcp 
- /tmp/dhcp.leases
- /tmp/hosts/odhcod

Also please share the output of "ip neigh" as well as the contents of the three files above.

anyuta1166, 2021-06-14:
> Also please share the output of "ip neigh" as well as the contents of the three files above.

```
root@margarita:~# cat /etc/config/dhcp

config dnsmasq
        option localise_queries '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option localservice '1'
        option ednspacket_max '1232'
        option domainneeded '1'
        option rebind_protection '0'
        option nonwildcard '0'
        option nonegcache '1'
        option cachesize '512'
        option expandhosts '1'
        option local '/lan/'
        option domain 'lan'
        option quietdhcp '1'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        option ra_maxinterval '600'
        option ra_mininterval '200'
        option ra_lifetime '1800'
        option ra_mtu '0'
        option ra_hoplimit '0'
        option ra_management '1'
        option limit '100'
        option leasetime '2h'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

root@margarita:~# cat /tmp/dhcp.leases
1623705199 80:ad:16:4f:0f:5c 192.168.20.159 Redmi4X-Redmi 01:80:ad:16:4f:0f:5c
1623703623 c6:e6:db:24:f7:24 192.168.20.130 Galaxy-A51-pol-zovatela-Viktoria 01:c6:e6:db:24:f7:24
1623705059 1c:bf:c0:c0:7d:e3 192.168.20.189 LAPTOP-L4DKFO9A 01:1c:bf:c0:c0:7d:e3
1623703514 1e:18:cd:8e:34:db 192.168.20.122 A51-pol-zovatela-Oksana 01:1e:18:cd:8e:34:db
1623704971 54:27:1e:b6:0f:58 192.168.20.120 PASHA 01:54:27:1e:b6:0f:58
1623706721 7c:1c:68:02:fa:5b 192.168.20.131 Galaxy-J5-2016 01:7c:1c:68:02:fa:5b
1623703771 a0:9d:c1:65:85:e3 192.168.20.199 * 01:a0:9d:c1:65:85:e3

root@margarita:~# cat /tmp/hosts/odhcpd
# br-lan 00010001285a4c16a8346a86aaaa 6a86aaaa - 1623698358 85e 128 fd87:90d1:5d0::85e/128

root@margarita:~# ip neigh
192.168.20.188 dev br-lan lladdr a8:34:6a:86:aa:aa STALE
192.168.20.199 dev br-lan lladdr a0:9d:c1:65:85:e3 REACHABLE
192.168.20.113 dev br-lan  FAILED
192.168.20.159 dev br-lan lladdr 80:ad:16:4f:0f:5c REACHABLE
192.168.20.189 dev br-lan  FAILED
192.168.20.111 dev br-lan  FAILED
192.168.20.130 dev br-lan lladdr c6:e6:db:24:f7:24 REACHABLE
192.168.20.122 dev br-lan lladdr 1e:18:cd:8e:34:db REACHABLE
192.168.20.106 dev br-lan  FAILED
192.168.20.117 dev br-lan lladdr 54:27:1e:d0:dc:3c STALE
192.168.20.120 dev br-lan lladdr 54:27:1e:b6:0f:58 REACHABLE
192.168.20.131 dev br-lan lladdr 7c:1c:68:02:fa:5b REACHABLE
192.168.20.185 dev br-lan  FAILED
fd87:90d1:5d0:0:aa34:6aff:fe86:aaaa dev br-lan lladdr a8:34:6a:86:aa:aa STALE
fd87:90d1:5d0:0:58be:7596:d7df:9d13 dev br-lan lladdr 7c:1c:68:02:fa:5b STALE
fe80::c4e6:dbff:fe24:f724 dev br-lan lladdr c6:e6:db:24:f7:24 DELAY
fe80::b53c:80fa:82f6:ab8c dev br-lan lladdr e4:aa:ea:5a:98:5f STALE
fd87:90d1:5d0:0:6ccc:740e:98ca:aed7 dev br-lan lladdr a0:9d:c1:65:85:e3 STALE
fe80::d562:b004:b9a7:1408 dev br-lan  FAILED
fe80::adb9:b57f:9e71:b087 dev br-lan lladdr 1c:bf:c0:c0:7d:e3 STALE
fd87:90d1:5d0:0:d5b:d4ab:c7a6:b519 dev br-lan lladdr 1c:bf:c0:c0:7d:e3 STALE
fd87:90d1:5d0:0:9065:58da:57c8:b2b2 dev br-lan lladdr 1c:bf:c0:c0:7d:e3 STALE
fd87:90d1:5d0:0:d8d6:5c67:c004:2a86 dev br-lan lladdr 80:ad:16:4f:0f:5c REACHABLE
fd87:90d1:5d0:0:90a4:41bb:4251:f8a7 dev br-lan lladdr e4:aa:ea:5a:98:5f STALE
fd87:90d1:5d0:0:575:1aa1:d5dd:8e28 dev br-lan lladdr 1c:bf:c0:c0:7d:e3 STALE
fe80::1c18:cdff:fe8e:34db dev br-lan lladdr 1e:18:cd:8e:34:db REACHABLE
fe80::aa34:6aff:fe86:aaaa dev br-lan lladdr a8:34:6a:86:aa:aa STALE
fe80::82ad:16ff:fe4f:f5c dev br-lan lladdr 80:ad:16:4f:0f:5c REACHABLE
fd87:90d1:5d0:0:24e4:9520:a89e:efe dev br-lan lladdr 1e:18:cd:8e:34:db REACHABLE
fe80::a29d:c1ff:fe65:85e3 dev br-lan lladdr a0:9d:c1:65:85:e3 STALE
fd87:90d1:5d0:0:a057:a37b:b9b4:67e2 dev br-lan  FAILED
fd87:90d1:5d0:0:3449:48d7:a989:e768 dev br-lan lladdr c0:8c:71:92:a6:fc router STALE
fe80::7e1c:68ff:fe02:fa5b dev br-lan lladdr 7c:1c:68:02:fa:5b REACHABLE
fe80::1037:7fff:fe06:9690 dev br-lan  FAILED
fd87:90d1:5d0:0:7c2c:e990:5f4b:b73c dev br-lan lladdr 12:37:7f:06:96:90 STALE
fd87:90d1:5d0:0:3964:a008:c12f:86a0 dev br-lan lladdr 3c:f0:11:a6:a1:35 STALE
fe80::c28c:71ff:fe92:a6fc dev br-lan lladdr c0:8c:71:92:a6:fc router STALE
fd87:90d1:5d0:0:54cf:52a4:6b70:95d9 dev br-lan lladdr 3c:f0:11:a6:a1:35 STALE
fd87:90d1:5d0:0:4145:2a66:379f:4656 dev br-lan lladdr c6:e6:db:24:f7:24 REACHABLE
```

> Does the behavior persist if you move any of the following files away?

Well, this Wi-Fi client has moved away so the "Associated Stations" list looks fine, but getHostHints still returns this
`{"12:37:7F:06:96:90":{"ipaddrs":["192.168.20.1","192.168.20.120","192.168.20.122","192.168.20.130","192.168.20.131","192.168.20.159"],"ip6addrs":["fd87:90d1:5d0:0:7c2c:e990:5f4b:b73c"],"name":"margarita.lan"}`

closed 2021-06-14 by jow-

jow-, 2021-06-14:
Fixes pushed to master and openwrt-21.02 branches. A recent refactoring of the host hints code introduced a logic flaw which led to the behavior you observed.

commit a78b7478 (2021-06-14): rpcd-mod-luci: refactor host hint gathering

 - Do not incorrectly put the IP addresses from the rrdns reply into the
   first hint, instead distribute hostnames to the appropriate hints.

 - Avoid returning stale lease information from previously parsed records
   by clearing the static struct before each parse attempt.

 - Improve existing checks for empty MAC addresses and unspecified IPv6
   addresses.

Fixes: #5126
Fixes: 87bcf9e923 ("rpcd-mod-luci: Return array of addresses in getHostHints")
commit e0a32b27 (2021-06-14): rpcd-mod-luci: refactor host hint gathering

 - Do not incorrectly put the IP addresses from the rrdns reply into the
   first hint, instead distribute hostnames to the appropriate hints.

 - Avoid returning stale lease information from previously parsed records
   by clearing the static struct before each parse attempt.

 - Improve existing checks for empty MAC addresses and unspecified IPv6
   addresses.

Fixes: #5126
Fixes: 87bcf9e923 ("rpcd-mod-luci: Return array of addresses in getHostHints")

(cherry picked from commit a78b74784dc1d4a4007e3e21a79228d5192f0dad)