openwrt/luci #5139: OLSRD: Changing LQ-Multiplikator is not properly applied (21.02-snapshot)

Noki, 2021-06-21:
When changing the OLSRD lq-multiplikator via Luci the changes are not immediately applied. Seems that olsrd is not automatically restarted when the config changes, but the changes are visible in the config file. Manually restarting olsrd / restarting the router solves solves the issue.

feckert, 2021-06-22:
I think the following code section is missing in the [olsrd init script](https://github.com/openwrt/routing/blob/master/olsrd/files/olsrd4.init).

```
service_triggers() {
        procd_add_reload_trigger "olsrd" "olsrd6"
}

```

@Noki Could you give theme a try? If this works, please open a P/R.
thanks

Noki, 2021-06-22:
I tested the trigger as proposed and can confirm that it work. @PolynomialDivision agreed to prepare a commit.

PolynomialDivision, 2021-06-22:
@feckert Thanks for taking care of that issue! :)

Noki, 2021-06-22:
@PolynomialDivision thanks for the commit!

PolynomialDivision, 2021-06-23:
Can be closed. ;)

closed 2021-06-23 by Noki